armbian/configng #5: Use (( )) instead of [[ ]]?

dimitry-ishenko, 2022-12-21:
(1) I see bash-utils heavily using
```bash
[[ $# -lt 2 ]] && ...
```
Wouldn't it be more readable to do
```bash
(( $# < 2 )) && ...
```
instead? `((` is used for arithmetic evaluation, which is what `-lt` is trying to force inside `[[`.

(2) Also, seeing this:
```bash
[[ $# = 0 ]] && ...
```
rubs me the wrong way on multiple levels.

First, we are all taught that `=` is assignment and `==` is equivalence comparison. Yes, in bash they are the same when used inside `[[`, but `=` is used for assignment elsewhere in bash and IMHO this is just a bad coding practice. 

Second, operator `=` within `[[` is used for pattern matching and not arithmetic comparison. So, it should minimally be:
```bash
[[ $# -eq 0 ]] && ...
```
or better yet:
```bash
(( $# == 0 )) && ...
```
to be more readable and consistent with (1).

dimitry-ishenko, 2022-12-21:
I've realized I should probably take it up with bash-utility. I don't see any activity in their repo for the past 2 years, but I've opened the same issue with them: https://github.com/labbots/bash-utility/issues/1

sgjava, 2022-12-21:
@dimitry-ishenko, knock yourself out if you can find a better bash library. I just didn't want to duplicate a lot of the boilerplate type stuff. I'm not sure how much bash really changes, thus maybe not a lot commits needed. On the plus side you can make all the Armbian stuff clean. I tried to go down the path of functional/procedural programming and not create God scripts or functions.

dimitry-ishenko, 2022-12-21:
@sgjava I am not saying the library is bad. I actually like it. But there quite a few things that can be improved. Ideally they would be done upstream, but if they don't respond, I am willing to fork it and do the clean-ups.

sgjava, 2022-12-21:
@dimitry-ishenko Yeah, exactly, that's why I embedded it into the project. I hate submodules since the project can change (or be deleted) and wreck your project. If you do go the submodule path just maybe create the fork under the Armbian project? I did research bash libraries out there and there aren't many. Anyways, it's all good. I'm mainly a Java, C, C++ and Python dev. I do script a lot with bash, but writing a complete project (or API) in it is more complicated than most folks think. Just missing real data types, etc. makes thing rough in my eyes.

dimitry-ishenko, 2022-12-21:
> If you do go the submodule path just maybe create the fork under the Armbian project?

IMHO this is the best approach. I don't have access to Armbian, so for now I've made my own fork here: 

https://github.com/dimitry-ishenko/bash-utility

If someone could fork the upstream project in Armbian and give me write access, I can work from there. Then it can be added here as a submodule.

FWIW, I've done some clean-ups in the Array module: https://github.com/dimitry-ishenko/bash-utility/blob/master/src/array.sh and submitted PR to upstream. Will see if they respond.

igorpecovnik, 2022-12-21:
> If someone could fork the upstream project in Armbian and give me write access, I can work from there. 

Invitation sent.

sgjava, 2022-12-22:
@dimitry-ishenko sure as long as you control the submodule. If you rely on a submodule you don't own and it's deleted then you are screwed. @igorpecovnik hooked you up with access.

Tearran, 2023-07-17:
Unwarranted use for readability: 

In Bash, (( .. )) is an arithmetic construct, while [[ .. ]] is a conditional expression. They serve different purposes and are used in different scenarios.

(( .. )) is used for arithmetic operations and comparisons. For example, you can use it to increment a variable ((i++)), or to compare two numbers ((a > b)).

On the other hand, [[ .. ]] is used for testing conditions. It is an improvement over the [ .. ] command and has several enhancements that make it a better choice if you write scripts that target Bash. 

For example, it handles empty strings and strings with whitespace more intuitively, it lets you use && and || operators for boolean tests and < and > for string comparisons, it has a =~ operator for doing regular expression matches, and it lets you do pattern matching 
[referance;1](https://superuser.com/questions/1533900/difference-between-and-or-and-in-bash) [referance;2](https://superuser.com/questions/1533900/difference-between-and-or-and-in-bash).

So in summary, (( .. )) is used for arithmetic operations while [[ .. ]] is used for testing conditions.

dimitry-ishenko, 2023-07-18:
> So in summary, (( .. )) is used for arithmetic operations...

You are contradicting yourself:

> (( .. )) is used for arithmetic operations and comparisons. For example, you can use it to increment a variable ((i++)), **or to compare two numbers ((a > b)).**

Unless you include comparison operations into arithmetic operations (which they actually are).

And, what I was pointing out was that bash-utility incorrectly uses `[[ ]]` to compare if the number of parameters passed to a function is less than 2 or equal to 0.

dimitry-ishenko, 2023-07-18:
BTW I do have a mostly cleaned up version of bash-util here: https://github.com/armbian/bash-util, which was supposed to become the submodule of this repo.

Unfortunately, I have not been able to complete it (due to lack of motivation). I am not sure at this point if it should be abandoned or it could still be added as a submodule and someone could finish cleaning it up concurrently.

Tearran, 2023-07-18:
Second, post.
Not Shure about submodules and upstream stuff,  but the lack of motivation I get.

closed 2023-12-06 by igorpecovnik