armbian/configng #12: Is it possible to have Freeze / Unfreeze in the menu?

igorpecovnik, 2023-12-06:
Try freezing unfreezing kernel in the previous armbian-config

Tearran, 2023-12-09:
Existing config features can be accessed through Legacy menu.

https://github.com/armbian/configng/assets/2831630/ab56d4f9-cdb3-4132-8011-41ae233c0d3d

Tearran, 2024-01-19:
Loads key=pairs form `/etc/armbian-release`

```
{
    source /etc/armbian-release
    packages=("linux-image-current-$LINUXFAMILY" "linux-u-boot-$BOARD-$BRANCH" "u-boot-tools")

    for pkg in "${packages[@]}"; do
        if apt-mark showhold | grep -q "^$pkg$"; then
            sudo apt-mark unhold "$pkg"
            echo "Unheld $pkg"
        else
            sudo apt-mark hold "$pkg"
            echo "Held $pkg"
        fi
    done
}
```

Tearran, 2024-05-23:
Reference code migrated from current armbian-config [Job](https://github.com/armbian/config/blob/master/debian-config-jobs#L947) to make
 [Helper Function  ](https://github.com/armbian/configng/blob/main/lib/armbian-configng/config.ng.functions.sh#L193)  
[Job Enable](https://github.com/armbian/configng/blob/main/lib/armbian-configng/config.ng.jobs.json#L8) and [Job Disable](https://github.com/armbian/configng/blob/main/lib/armbian-configng/config.ng.jobs.json#L8)

Tearran, 2024-05-23:
Reference code migrated from current armbian-config [Job](https://github.com/armbian/config/blob/master/debian-config-jobs#L947) to make
[Helper Function](https://github.com/armbian/configng/blob/main/lib/armbian-configng/config.ng.functions.sh#L193)
[Job Enable](https://github.com/armbian/configng/blob/main/lib/armbian-configng/config.ng.jobs.json#L8) and [Job Disable](https://github.com/armbian/configng/blob/main/lib/armbian-configng/config.ng.jobs.json#L8)