armbian/configng #29: Questions/thoughts around a framework for configng.

schwar3kat, 2023-12-13:
"About
Next generation bash based configuration API"
Configng doesn't appear to be an API. Is it an API?  Is it perhaps a bash based menu driven configuration utility or something else?

Some thoughts/questions about a framework:
First...
Should the framework support multiple languages by design for all displayed text, menu's, and all dialogs? Is there any other Armbian project doing this well that can be used as a template? _See https://github.com/armbian/configng/issues/26#issuecomment-1853614743_
Should the framework promote documentation as a required development element? Is there any other Armbian project doing this well that can be used as a template?
Should the framework require logical defined naming conventions for functions and variables? Is there any other Armbian project doing this well that can be used as a template?
Should the framework include any elements promoting accessibility, if so which and how?

Then...
What kind of structure should menu system have (scalable nesting/infinite nesting/limited nesting)? 
What format should the menu configuration be? (some ideas exist. in the light of the above, are they suitable as a starting point).
What kind of structure should menu driven functions have? (some development exists. in the light of the above, is it suitable as a starting point).
What kind of extensible structure should reusable code elements have (functions, board variables and etc. Modular/monolithic. Format, location)? For context, armbian-config has these. Can we do better?

schwar3kat, 2023-12-21:
Developer documentation should possibly be included in the code, and could possibly be exported to be accessible from and formatted for Github.   It should describe how the function works, it's structure, it's interactions and requirements.  This can possibly be enabled to some extent with a form in the header and or comments.  The key here, I think, would be to provide a framework for developers to fill in, attached to each function.

In my opinion, end user documentation should probably be geared towards providing end user instructions in a format compatible with https://docs.armbian.com/User-Guide_Armbian-Config/ for inclusion into that website.  This can be done via https://github.com/armbian/documentation.

Each menu function could have a thorough description and illustrated user instructions, preferably with screen shots.  Anything auto-generated, will almost certainly require editing and proof-reading before inclusion.  Some means of ensuring that it happens before the change PR is accepted would be good.   Perhaps Github workflow.

Tearran, 2023-12-23:
The configng is meant to adhere to GNU Bash standard of limitations layout by its community and by common best [practises](https://devhints.io/bash). 
These limitations are not arbitrary. They are based on the experience of the bash community, and they are designed to prevent bash from being used to harm systems.

The procedures and request of https://armbian.atlassian.net/browse/AR-967 do not indicate any acknowledgment any bash limitations. And a few confusing concepts such as a bash gui not impossible but really? To overcome this a compromise was made to allow for the foundation to follow the standards, while providing some code flexibility for work-arounds, the libraries does NOT adhered to the GNU bash standers.   

A clear establishment of limitations and expectations of responsibility. Example of how to approach are not easily identified in the threads. Understandably the forum thread focus is more about overcoming known limitation or issue. Not establishing them. 

The term  "Documentation" I have been using as general term. At this stage is not currently able to offer complete entries for https://docs.armbian.com/User-Guide_Armbian-Config/ 

The time , resource required to research and test an integrations of https://squidfunk.github.io/mkdocs-material/ library was not available a compromise was made that included the basics vanilla procedure to build integration off of later.
That is also used for keeping track of progression and digression until time or expertise is available. was made

- [Help messages](https://github.com/armbian/configng/blob/main/bin/armbian-configng#L134)
- Parsing the command to the function. 
- json, html, css etc
- early test
    - [readme.md](https://github.com/Tearran/configng/blob/main-testin/docs/mardown/armbian-config.1.md). Man Page format
    - [man pages](https://github.com/Tearran/configng/tree/main-testin/share/man),   troff converted from above readme

Tearran, 2023-12-26:
Context Refinement:

Doc-Gens and Complexity:

Observation:
Leveraging doc-gens for contributors' assistive devices may complicate the project beyond practical use.

Consideration:
A modular approach was initially considered. A doc-gen is needed, and a refinement of usable information for Armbian use is also clearly needed.

Option:
Limit the doc-gen to CSV and Markdown initially. Address leveraging other conversions later.

Rationale:

A modular approach aligns with the project's adaptability.
Focusing on CSV and Markdown caters to immediate documentation needs.
Gradual incorporation of additional formats maintains a manageable complexity.
Objective:
Efficiently integrate doc-gens to facilitate contributors' understanding, keeping the documentation practical and adaptable for various use cases within the Armbian environment.

Contextual Inquiries:

Developer Knowledge:

Current Assumption: Not all developers are expected to inherently know the location and roles of /etc, /var, and /share and how they relate to Armbian.
Observation: Armbian Build developers are likely to possess this knowledge.
Exposing Information:

Observation: UX needs to know where and what settings to read, while the backend needs to know where and what to set.
Question: What information should be exposed, and what need not be?
Roles and Responsibilities:

Question: How can roles and responsibilities be clearly defined before delving into code solutions?
Observation: There might be a perception that clearly defining roles introduces unnecessary complexity.
Doc-Gens and Complexity:

Observation: Leveraging doc-gens for contributors' assistive devices may complicate the project beyond practical use.
Consideration: A modular approach was initially considered and can be revisited later. The focus should be on testing and replicating documentation.
Separation of Concerns:

Question: If the /lib group function were in the existing config repository and configng remained in this repository, could that offer context to the outlined separation of responsibility and expectations?
TUI Integration:

Question: Is the project's focus on clearly defining roles being interpreted as unnecessary complexity?
Observation: Dialog and whiptail introduce complexity to parsing and nesting arrays, potentially acting as bottlenecks.
Temporary TUI Replacement:

Option: Temporarily replace the Dialog and Whiptail requirement with a simple bash read, utilizing tput for background and foreground colors.
Observation: This allows for a quick fix, focusing on nesting placeholders while exploring other TUI/gui options. Example with armbian-configng -t.
Conclusion:

The project's concept evolution emphasizes adaptability, responsiveness to user and developer needsThat are only assumed to this point, and a practical approach to complexity. The ongoing focus is on achieving a balance between clarity, user-friendliness, and efficient functionality. The outlined contextual inquiries guide the project's trajectory, ensuring that each decision aligns with its overarching goals.

closed 2024-09-02 by Tearran