armbian/configng #53: Propose small design change

igorpecovnik, 2024-09-04:
When we are running in CLI mode it would be desirable to have a switch to execute command without confirmation prompt.

Example:
`sudo bin/armbian-configng --cli S11`

This switch to stable repository and we can skip the prompt in machine mode. One way it would be to move this from command to new field

"prompt": [ "This will open interface file to edit\nCTRL+S to save\nCTLR+X to exit\nwould you like to continue?"

which we don't execute in CLI mode.

Tearran, 2024-09-06:
What is the purpose of CLI mode?

I assumed it is for CI/CD, GitHub Actions, or script integration. If this assumption is correct, then the --api flag is appropriate.

The current --cli flag should then be called --tui since it just bypasses menu selections, and --api should be invoked with --cli.

Tearran, 2024-09-09:
>This switch to stable repository and we can skip the prompt in machine mode. One way it would be to move this from command to new field

good idea
 implemented prompt 

example:
```json
"prompt": "This will open /boot/armbianEnv.txt file to edit\nCTRL+S to save\nCTLR+X to exit\nwould you like to continue?",
```

>When we are running in CLI mode it would be desirable to have a switch to execute command without confirmation prompt.
>
>Example:
>sudo bin/armbian-configng --cli S11

The only issue I see here is `--cli` flag may be misleading, `--cli` is `TUI` or `simple mode` 

 --api is intended for "cli mode" or "advanced mode"  passing command line arguments in a direct manner to avoid the nesting involved with the UI

igorpecovnik, 2024-09-09:
> flag may be misleading

Yes ofc. What about: --cmd or nothing? just -S11 ?

closed 2024-09-10 by igorpecovnik