armbian/configng #117: [Task]: Merge status and disabled into one functionality

igorpecovnik, 2024-09-26:
### Task description

JSON has status and disabled ( to not show in the menu), but this can be done via "staus". @gu

viraniac, 2024-09-27:
I don't know what status is for, but I don't think it was ever used for hiding menu entries. I thought its for documentation purpose only. 

Previously there was a field called `show` that was being used everywhere to show and hide menu items. Once I got conditions working, there was no need to set show as true in every json entry and hence I removed it. There is no need for `disabled` either, the only reason its there is because configng has dummy entries like [this](https://github.com/armbian/configng/blob/bd256bc4f9bbf29e699054c991a6ea11a450fd67/lib/armbian-configng/config.ng.jobs.json#L525). Once such entries are removed, `disabled` can be removed as well. I had no clue if those entries were serving any purpose or not, so I left them alone

Probably @Tearran can comment better as he was involved from the start

Tearran, 2024-09-27:
although extends the purpose does seem to fix well.
"status": "disabled"

closed 2024-09-30 by igorpecovnik