armbian/configng #161: [Feature Request]: Store and show port number

igorpecovnik, 2024-10-08:
### Which feature would you like to have?

Several software packages will have service running at particular ports and we need to show that information somehow in the menu it self. For example - Cockpit is using port 9090

[image]

Perhaps M03 - Stop Cockpit service (http://IP_ADDDRESS:9090) ?

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

Tearran, 2024-10-11:
>NOTE:

The Cockpit installation sets up a MOTD, allowing us to retrieve service port information
```
cat /run/cockpit/motd | sed 's/Web console: //'
```

igorpecovnik, 2024-10-11:
> The Cockpit installation sets up a MOTD

Yes, I have noticed that. Some do this, some not.

closed 2024-10-13 by Tearran