armbian/configng #221: Cannot access Portainer web GUI

nkev, 2024-11-02:
### What happened?

`SBC`: Orange PI 5B 
`OS`: Armbian 24.8.1 Bookworm Minimal / IOT Kernel: 6.1.75, Size: 230 MB, Release date: Aug 24, 2024
`OS Disk`: Sandisk Ultra 16GB

The Portainer installation was successful (at least no error), but I cannot access the web GUI from my Macbook Pro

### How to reproduce?

1. First install configng:

```BASH
echo "deb [signed-by=/usr/share/keyrings/armbian.gpg] https://armbian.github.io/configng stable main" | sudo tee /etc/apt/sources.list.d/armbian-development.list > /dev/null
sudo apt update
sudo apt -y install armbian-config
```

2. Then run `armbian-config` and install Portainer:
```BASH
portainer_data
Unable to find image 'portainer/portainer-ce:latest' locally
latest: Pulling from portainer/portainer-ce
2a8c27161aa3: Pull complete 
679061c2c821: Pull complete 
39b5457baa4a: Pull complete 
27fb5b6e87d5: Pull complete 
ad26f2495f3e: Pull complete 
25275c19b336: Pull complete 
683befaed514: Pull complete 
888d92095067: Pull complete 
01a2982e5693: Pull complete 
2ff32abeca46: Pull complete 
4f4fb700ef54: Pull complete 
Digest: sha256:f2a7f5abd4735f9cd91563c6134e014b15168c4018beea87f1eec9d9618b2ad4
Status: Downloaded newer image for portainer/portainer-ce:latest
3716121411f83e8529c3b29de698ee9d2cd17241afa287e32d7f5a44ded4a23d
Armbian Config: The Next Generation
```
3. Confirm Portainer is running on Orange Pi 5B:

```BASH
root@orangepi5:~# docker ps
CONTAINER ID   IMAGE                    COMMAND        CREATED         STATUS         PORTS                                                             NAMES
3716121411f8   portainer/portainer-ce   "/portainer"   5 minutes ago   Up 4 minutes   8000/tcp, 9443/tcp, 0.0.0.0:9002->9000/tcp, [::]:9002->9000/tcp   portainer
```

4. Use Chrome on my Macbook Pro to access Portainer running on Orange PI 5B via `http://192.168.0.69:9443/` returns:

`This site can’t be reached. 192.168.63.69 refused to connect.`

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-11-07:
> The only one that displays a webpage is https://192.168.0.69:9002/

Don't understand this questions well, but this is expected behaviour.

https://github.com/armbian/configng/blob/main/tools/modules/software/install_portainer.sh#L23

https://medium.com/@maheshwar.ramkrushna/chap-11-docker-port-mapping-exposing-container-services-to-the-host-system-66de32abdb2f

nkev, 2024-11-07:
The issue is in step 4 above.
If you try the steps above on a Macbook, you will not be able to access Portainer on ports 9443, 8000, 9000 so something is wrong in the installation script.

Tearran, 2024-11-08:
Replicated, remotely on windows chrome browser. 
Not a bug, this is expected behaviour, So is possibly a TL;DR user error. 

 https://portal.portainer.io/knowledge/your-portainer-instance-has-timed-out-for-security-purposes

fix
```
 sudo docker stop portainer
 ```
then
```
 sudo docker start portainer
```

[image]

Perhaps to help prevent this type of user error, I suggest A feature request to have start, stop, restart in the module

nkev, 2024-11-08:
@Tearran I gave you the exact steps to reproduce the error. If you replicated it then you saw that you need to restart the docker container, which is an installation issue with your script, not a "TL;DR user error".

Tearran, 2024-11-09:
@nkev I am here to help, please clarify Is this a bug report or a feature request.

if this is bug report - there are no unexpected behavior or errors works as expected. this issue can be closed. 

if this is a feature request - Hey use the Portainer: default ports 

I do not use this tool so only assuming, changing the port from 9002 to 9000 and add 9443 should work.

>Note: tested: 
successful login page on port 9000 changed module line to
https://github.com/armbian/configng/blob/main/tools/modules/software/install_portainer.sh#L23
``` bash
docker run -d -p '9000:9000' --name=portainer --restart=always \
```
>Note: untested:
https://github.com/armbian/configng/blob/main/tools/modules/software/install_portainer.sh#L23 source google redit and example above
the sugeted command for ports 9000 & 9443 

``` bash
docker run -d -p '9000:9000' -p '9443:9443' --name=portainer --restart=always \
```

igorpecovnik, 2024-11-28:
@nkev This should be fixed now, update package and try again. Closing.

closed 2024-11-28 by igorpecovnik