armbian/configng #222: [Bug]: SY008 - Change shell system wide to BASH  throws error

schmiedelm, 2024-11-05:
### What happened?

I am starting with an Image with the zsh Shell active and trying to change the shell with the menu entry SY008.

Logged in as a user (not root)

After the command was running, nothing changed in the menu, and after leaving armbain config I see.

```
Password: chsh: PAM: Authentication failure
Password: chsh: PAM: Authentication failure
```

### How to reproduce?

- use Image with zsh shell active
- log in as user
- run configng as root
- choose SY008 - Change shell system wide to BASH

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

schmiedelm, 2024-11-05:
The link to the Code of Conduct in the template for the Bug Report doesn't work.

igorpecovnik, 2024-11-05:
Reproduced.

closed 2024-11-06 by igorpecovnik