armbian/configng #242: [Task]: Interesting and valuable titles / features to add to configurator

igorpecovnik, 2024-11-10:
### Task description

- [ ] Kodi
- [ ] Chromium (port from MESA extension)
- [ ] GoogleChrome (we host at our repo)
- [ ] Firefox (we host non snap version at our repo)
- [ ] LAMP stack with CERT bot
- [ ] Samba Server (enable, disable, config)
- [ ] NFS Client (enable, disable, config)
- [ ] Nextcloud
- [ ] XRDP
- [ ] proFTPD
- [ ] Home Assistant (taking from here https://github.com/armbian/configng/pull/235/files)
- [ ] Syncthing
- [ ] WireGuard
- [ ] Sonarr
- [ ] Radarr
- [ ] Grafana
- [ ] OctoPrint
- [ ] VSCodium (hosted at our repo)
- [ ] VSCode (hosted at our repo)

dimitry-ishenko, 2024-11-11:
If nobody is working on Samba, I can give it try.

EvilOlaf, 2024-11-14:
Jellyfin might be interesting, especially for rk3588 based devices since they actually support this platform.

EvilOlaf, 2024-11-19:
My observations when setting up atuin server and testing:  

After initial installation the config file isn't written directly but is created after `register`.  
When registering with `ATUIN_SYNC_ADDRESS=https://services.armbian.de/atuin` this value is not written back in the freshly generated config and needed adjustment.  
Therefore when implementing this via configng the config file with adjustments needs to be written the installation (line 32: `sync_address = "https://services.armbian.de/atuin"`)

EvilOlaf, 2024-11-27:
What about wirelogd?
https://github.com/nikaro/wirelogd

igorpecovnik, 2024-11-27:
> What about wirelogd?

Yes. First step - adding to our repo https://github.com/armbian/os/blob/main/external/gh.conf

> don't have permission.

Try now
 
> *rr stack is WIP

This is ready, tested, all works, won't be adding anything more there for this section.

> I suggest to remove this and add as module for configng:

Agree. There is also this as an option https://docs.linuxserver.io/images/docker-wireguard/?h=wire

igorpecovnik, 2025-01-16:
Yes, I saw. I think it can be valuable, lets add it.

igorpecovnik, 2025-01-29:
> If nobody is working on Samba, I can give it try.

Did you have time to look into this? Samba is pretty high on the usage interest level.

dimitry-ishenko, 2025-01-29:
> > If nobody is working on Samba, I can give it try.
> 
> Did you have time to look into this? Samba is pretty high on the usage interest level.

Sorry, got side-tracked with other things. Will try to wrap it in a few days

Tearran, 2025-04-16:
It seems I unintentionally blur the lines between creating a "proof of concept" and developing a "practical application." While my goal is to focus on reusable UX/UI modules and helpers (at this time) (as seen in [PR #529](https://github.com/armbian/configng/pull/529) and [PR #530](https://github.com/armbian/configng/pull/530)), I often find the proof of concept evolving into something more functional than intended.

This brings me to a question for the community:
Would it benefit armbian-config to separate package types (e.g., deb, containerized, git/zip/tar) to better organize and manage the growing list of titles? [apt-get/deb](https://github.com/armbian/configng/pull/530) 

This idea stems from a reusable UX/UI checklist concept, which could provide transparency and improve usability by grouping packages based on their source or installation method. However, my primary focus remains on refining the UX/UI foundation rather than the practical applications of the proof of concept.

I’d appreciate input on:

Whether separating package types would be beneficial.
How to balance UX/UI refinement with potential practical applications.

Tearran, 2025-04-18:
### Following Software titles used the frameworks pkg_* to install/remove 
-  proFTPD  (frameworks pkg_* version)
    - Does this differ from  (we host at our repo)
-  Firefox ESR (frameworks pkg_* version) https://github.com/armbian/configng/pull/534
    - Does this differ from  (we host at our repo)
-  Chromium (frameworks pkg_* version) https://github.com/armbian/configng/pull/534
    - Does this differ Chromium ([port from MESA extension](https://github.com/armbian/build/blob/main/extensions/mesa-vpu.sh))
-  Lynx command line web browser (frameworks pkg_* version) https://github.com/armbian/configng/pull/534
    - Text web browser for IOT/MIN images.