armbian/configng #243: armbian-config allwinner directory no longer in /boot/dtb

stephengraf, 2024-11-10:
### What happened?

Ttrying to manage overlays in armbian-config produces error message:

ls: cannot access '/boot/dtb/allwinner/overlay/*.dtbo': No such file or directory
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.

### How to reproduce?

Testing oranpione debian current for Nov release.
v24.11 rolling for Orange Pi One running Armbian Linux 6.6.54-current-sunxi

The overlay directory is now in /boot/dtb. Directory /file/boot/allwinner no longer exists.

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-11-10:
> The overlay directory is now in /boot/dtb. Directory /file/boot/allwinner no longer exists.

Hmm, but all others have /boot/vendor/overlay ... do you know when was this changed and why?

igorpecovnik, 2024-11-10:
Perhaps @The-going or @pyavitz can tell more about this.

stephengraf, 2024-11-11:
>     The overlay directory is now in /boot/dtb. Directory /boot/dtb/allwinner no longer exists.

I am sure that this has changed since the August release as I did not experience this problem when doing the release testing then.
The overlay directory used to be in /boot/dtb/allwinner/overlay where the current script expects it to be.
The error is occurring in legacy, current and edge builds.

The-going, 2024-11-11:
> Perhaps @The-going or @pyavitz can tell more about this.

Everything is present in my test builds for linux-dtb-* version 24.8.0.
As `/boot/dtb-6.6.43-current-sunxi/*.dtb` `/boot/dtb-6.6.43-current-sunxi/overlays/{*.dtbo,*-fixup.scr,README*}`
For version 24.11.0, all files (*.dtb and*.dtbo) are in the same folder.
As `/boot/dtb-6.10.9-edge-sunxi/{*.dtb,*.dtbo}` only.

**This is a bug in the build script.**
But not armbain-config.
Forum: [breake-dtb-overlay](https://forum.armbian.com/topic/45546-6109-breake-dtb-overlay/)

although there is BASH code in some places here, which can in some cases generate ambiguous results.
As `VAR=$(ls /path/to/folder)`. [manage_dtoverlays.sh#L20](https://github.com/armbian/configng/blob/d70cdbf76fc7d68f97366ae1d00ead9d91e38ae4/tools/modules/system/manage_dtoverlays.sh#L20)
It's just grumbling about the coding style.

The-going, 2024-11-11:
> ### How to reproduce?
> 
> Testing oranpione debian current for Nov release. v24.11 rolling for Orange Pi One running Armbian Linux 6.6.54-current-sunxi

If this is the case, then the changes occurred in the kernel build script between v24.8 and v24.11.
The current patches affecting the package build do not seem to have changed.

In any case, it requires attention.

igorpecovnik, 2024-11-11:
> Please open the topic. I will continue.

done.

> This is a bug in the build script.

Agree. This bug has to be fixed prior releasing images.

The-going, 2024-11-11:
> > This is a bug in the build script.
> 
> Agree. This bug has to be fixed prior releasing images.

I started checking it out.
The beginning of compilation showed that the `dts` files and `dtso` overlays are compiled.

The-going, 2024-11-11:
Let's continue here: [7456](https://github.com/armbian/build/issues/7456)

igorpecovnik, 2024-11-11:
OK. Closing as its unrelated to armbian-config.

closed 2024-11-11 by igorpecovnik

igorpecovnik, 2024-11-13:
> Where is the OVERLAY_DIR variable initialized?

```
grep -R OVERLAY_DIR
sources/families/k3.conf:declare -g OVERLAY_DIR="/boot/dtb/ti/overlay"
sources/families/mvebu64.conf:OVERLAY_DIR="/boot/dtb/marvell/overlay"
sources/families/bcm2711.conf:declare -g OVERLAY_DIR="/boot/dtb/broadcom/overlay"
sources/families/include/sunxi64_common.inc:declare -g OVERLAY_DIR="/boot/dtb/allwinner/overlay"
sources/families/include/meson64_common.inc:OVERLAY_DIR="/boot/dtb/amlogic/overlay"
sources/families/include/rockchip64_common.inc:OVERLAY_DIR="/boot/dtb/rockchip/overlay"
sources/families/include/sunxi_common.inc:declare -g OVERLAY_DIR="/boot/dtb/allwinner/overlay"
sources/families/include/imx8_common.inc:OVERLAY_DIR="/boot/dtb/freescale/overlay"
sources/families/sm8250.conf:declare -g OVERLAY_DIR="/boot/dtb/qcom/overlay"
```
This is then save to BSP package config file that is always read at armbian-config:

```
cat /etc/armbian-release 
# PLEASE DO NOT EDIT THIS FILE
BOARD=zeropi
BOARD_NAME="ZeroPi"
BOARDFAMILY=sun8i
BUILD_REPOSITORY_URL=git@github.com:armbian/build.git
BUILD_REPOSITORY_COMMIT=45a1a09e9
LINUXFAMILY=sunxi
ARCH=arm
BOOT_SOC=
IMAGE_TYPE=user-built
BOARD_TYPE=csc
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage
KERNEL_TARGET=legacy,current,edge
KERNEL_TEST_TARGET=legacy
FORCE_BOOTSCRIPT_UPDATE=
FORCE_UBOOT_UPDATE=
OVERLAY_DIR="/boot/dtb/allwinner/overlay"
```

The-going, 2024-11-13:
If the package contains the correct path to the overlays as:
```
leo@bananapim3:~$ ls -al /boot/dtb
lrwxrwxrwx 1 root root 31 ноя 12 23:48 /boot/dtb -> dtb-6.11.6-edge-sunxi/allwinner

leo@bananapim3:~$ ls -al /boot/dtb/
итого 5348
drwxr-xr-x 3 root root 12288 ноя 12 22:25 .
drwxr-xr-x 3 root root  4096 ноя 12 22:25 ..
drwxr-xr-x 2 root root  4096 ноя 12 22:25 overlay
-rw-r--r-- 1 root root 33957 ноя 12 21:51 sun4i-a10-a1000.dtb
-rw-r--r-- 1 root root 30725 ноя 12 21:51 sun4i-a10-ba10-tvbox.dtb
-rw-r--r-- 1 root root 31183 ноя 12 21:51 sun4i-a10-chuwi-v7-cw0825.dtb
-rw-r--r-- 1 root root 33693 ноя 12 21:51 sun4i-a10-cubieboard.dtb
....
```
Pay attention.
The symbolic link was fixed manually after installing the package, as the package `postinst` script created a broken link.

But at the same time, the overlay folder does not contain an overlay file for this processor,
I get the following message in the console:
```
ls: unable to access '/*.dtbo': There is no such file or directory
Usage: grep [PARAMETER]... TEMPLATES [FILE]…
Run "grep --help" to get a more detailed description.
```
[image]
[image]

At the same time, it is impossible to exit the dialogue.
The <ctrl>\<c> key combination does not work.

The-going, 2024-11-13:
> This is then save to BSP package config file that is always read at armbian-config:

```
leo@bananapim3:~$ cat /etc/armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=bananapim3
BOARD_NAME="Banana Pi M3"
BOARDFAMILY=sun8i
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=0536db595a
LINUXFAMILY=sunxi
ARCH=arm
BOOT_SOC=
IMAGE_TYPE=user-built
BOARD_TYPE=csc
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage
FORCE_BOOTSCRIPT_UPDATE=
FORCE_UBOOT_UPDATE=
VENDOR="Armbian_community"
VENDORDOCS="https://docs.armbian.com"
VENDORURL="https://github.com/armbian/build"
VENDORSUPPORT="https://community.armbian.com/"
VENDORBUGS="https://github.com/armbian/community/issues"
BOOTSCRIPT_FORCE_UPDATE="no"
BOOTSCRIPT_DST="boot.cmd"
VERSION=24.8.0-trunk
REVISION=24.8.0-trunk
BRANCH=edge
```

Conclusion: The armbian-config package must have a dependency on the BSP package.

The-going, 2024-11-13:
> ```
> OVERLAY_DIR="/boot/dtb/allwinner/overlay"
> ```
> 
> 
>     
>   
> 
> For consistency reasons, we should always keep /boot/dtb/soc_key/overlay

the package `postinst` script created the `/boot/dtb -> dtb-6.11.6-edge-sunxi` symbolic link

igorpecovnik, 2024-11-13:
> we should always keep /boot/dtb/soc_key/overlay

in the package we should have dtb-6.11.6-edge-sunxi/allwiner/overlay ... dir symlinking is ok.

The-going, 2024-11-13:
> I want to keep armbian-config as general as possible so implementing a check to not show the menu entry if this variable is not set ... is possible solution.

In case of any script failure, the best solution is to display a menu with an error message and exit buttons and go back, or if the error is handled in the script itself, then the script message.

> in the package we should have dtb-6.11.6-edge-sunxi/allwiner/overlay ... dir symlinking is ok.

In this case, it is necessary to change the "postinst" script of the "linux-dtb" package in the build system,
which will affect everyone. It will take us a long time to coordinate.

igorpecovnik, 2024-11-13:
> In this case, it is necessary to change the "postinst" script of the "linux-dtb" package in the build system,
which will affect everyone. It will take us a long time to coordinate.

I don't understand why only Allwinner has this problem?

The-going, 2024-11-13:
> > In this case, it is necessary to change the "postinst" script of the "linux-dtb" package in the build system,
> > which will affect everyone. It will take us a long time to coordinate.
> 
> I don't understand why only Allwinner has this problem?

Maybe I have the reason locally on my device.
It seems that "u-boot" is looking for the dtb file in the wrong place.
Check