armbian/configng #248: [Bug]: No rolling release?

dimitry-ishenko, 2024-11-12:
### What happened?

I get the following error after switching to rolling release:
```
Hit:1 http://archive.ubuntu.com/ubuntu noble InRelease
Hit:3 http://archive.ubuntu.com/ubuntu noble-security InRelease
Hit:4 https://armbian.github.io/configng stable InRelease
Hit:5 http://archive.ubuntu.com/ubuntu noble-updates InRelease
Hit:6 http://archive.ubuntu.com/ubuntu noble-backports InRelease
Ign:2 http://fi.mirror.armbian.de/beta noble InRelease 
Err:7 http://netcup.armbian.com/beta noble Release     
  404  Not Found [IP: 152.53.15.45 80]
Reading package lists... Done
E: The repository 'http://beta.armbian.com noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Checking the http://beta.armbian.com link, it's missing the needed dirs. Is there such a thing as the rolling release?

### How to reproduce?

```
sudo armbian-config --cmd SY010
```

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-11-12:
Yes Beta repo needs to be rebuilt. Though this has nothing to do with config.

igorpecovnik, 2024-11-12:
This happens sometimes - bug in CI automation.

Tearran, 2024-11-12:
Context: BETA releases are generally unstable, and occasional failures should be expected. If Issues with beta.armbian.com arise thy are usually resolved in a reasonable time by their respective teams.

@dimitry-ishenko Thanks for the clear report. You made it easy to replicate, It is clear now this is not an armbian-config bug.

However, this does raise the need for some additional error handling.

Perhaps we could discuss implementing a message like, "Servers are unavailable at this time. Please try again later." 

In your opinion would such a message help prevent similar bug reports in the future?

dimitry-ishenko, 2024-11-12:
I thought that maybe this was an old option and beta no longer exists, which is why I've filed it here. If that's not the case, I am happy to close the issue.

@Tearran IMHO there is no need to give a message. The error in my first post should be sufficient for most users.

closed 2024-11-13 by Tearran