armbian/configng #250: [Bug]: [Manage device tree overlays ] does not works

XGFan, 2024-11-13:
### What happened?

There are no environment variables OVERLAY_DIR and BOOT_SOC when armbian-config is running.

### How to reproduce?

Fresh install on rock5c
OS: Fresh install on rock5c, Ubuntu Noble

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-11-13:
> There are no environment variables OVERLAY_DIR and BOOT_SOC when armbian-config is running.

How we can recreate this? Which image are you using? Self-build v24.11 or nightly should have this.

igorpecovnik, 2024-11-13:
Tried now on Orangepi 5 nightly builds. Works as expected. Don't have Rock5C on hands right now:
[image]

[image]

igorpecovnik, 2024-11-19:
Thanks for feedback. Yes, we are now updating directly from development, so fixes should go into production faster.