armbian/configng #265: [Bug]: unknown compression for member 'control.tar.zst'

way5, 2024-11-20:
### What happened?

```bash                
dpkg-deb: error: archive '/var/cache/apt/archives/armbian-config_25.2.0-trunk.47.1118.163006_all.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /var/cache/apt/archives/armbian-config_25.2.0-trunk.47.1118.163006_all.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /var/cache/apt/archives/armbian-config_25.2.0-trunk.47.1118.163006_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### How to reproduce?

Hey everyone!
I did the following:

```bash
sudo apt -y autoremove && sudo apt clean && sudo apt -y purge ?config-files
apt update && sudo apt -y full-upgrade
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org bullseye-security InRelease
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease                            
Hit:4 http://deb.debian.org/debian bullseye-backports InRelease                          
Hit:6 https://github.armbian.com/configng stable InRelease                               
Hit:5 http://armbian.tnahosting.net/apt bullseye InRelease
Reading package lists... Done                             
Building dependency tree... Done
Reading state information... Done
5 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  armbian-bsp-cli-radxa-zero armbian-firmware linux-dtb-current-meson64
  linux-image-current-meson64
The following packages will be upgraded:
  armbian-config
1 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Need to get 114 kB of archives.
After this operation, 612 kB disk space will be freed.
Get:1 https://github.armbian.com/configng stable/main arm64 armbian-config all 25.2.0-trunk.47.1118.163006 [114 kB]
Fetched 114 kB in 1s (134 kB/s)                       
dpkg-deb: error: archive '/var/cache/apt/archives/armbian-config_25.2.0-trunk.47.1118.163006_all.deb' uses unknown compression for member 'control.tar.zst', giving up
dpkg: error processing archive /var/cache/apt/archives/armbian-config_25.2.0-trunk.47.1118.163006_all.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /var/cache/apt/archives/armbian-config_25.2.0-trunk.47.1118.163006_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I know the compression method has changed but how to install in on bullseye now anyways? 
Thanks for any advice.

```bash
apt search zstd
zstd/oldstable,now 1.4.8+dfsg-2.1 arm64 [installed]
  fast lossless compression algorithm -- CLI tool

cat /etc/os-release
PRETTY_NAME="Armbian 24.8.4 bullseye"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.armbian.com/"
SUPPORT_URL="https://forum.armbian.com"
BUG_REPORT_URL="https://www.armbian.com/bugs"
ARMBIAN_PRETTY_NAME="Armbian 24.8.4 bullseye"
```

### On which host OS are you running the build script and observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-11-20:
This problem is related to packaging / APT. I already looked yesterday if there is a switch in rerepro to change compression mechanism, but that is not the case. There is no simple solution. Debian Bullseye is in theory still supported upstream, while we had to ditch it due to [raising compatibility issues](https://github.com/armbian/build/commits/main/config/distributions/bullseye/support) nobody (upstream) addresses. Our primary focus is arm and arm64 while at least armhf is getting less and less attention. If Debian doesn't take care of the packaging, there is nothing we can do. Providing support for stable packages is already on the edge of possible. 

I'll keep the issue open, as perhaps someone has a solution for this, but I strongly suggest you to upgrade to Bookworm. The problem will be gone there.

igorpecovnik, 2024-11-28:
Fixed with https://github.com/armbian/scripts/commit/9fe944b9c275116a1811f24c76714846ced39f50

closed 2024-11-28 by igorpecovnik