armbian/configng #266: [Task]: Collections of small future improvements

igorpecovnik, 2024-11-22:
### Task description

Let's assemble a list first, coding starts when this note is removed.

- [ ] Show actual IP address in the dialog box after installing applications. Currently it shows: `https://<your.IP>:8123`
- [ ] 
- [ ] 
- [ ]

Tearran, 2024-11-22:
>note: The distinction from `interactive mode` `non-interactive mode`. Was somewhat lost with the divergent keys of the json.  modules are built around module_options `database`

The database need attention. the keys are outdated such as "feature" was replaced with "id" in the json object and other keys were added. 

Ideally Before adding to the `Database` it need to be updated to reflect the clearer meaning `json` `keys`. 

### basic/interactive mode
`--cmd` previously `--cli` , user testing and a shortcut to the menu title 
`armbian-config --cmd  Benchy`
`armbian-config --cmd MAN005`

### advanced/non-interactive mode
`--api` option for `unit testing` modules, outputting` db info` into `json, md, config/` etc. 
`armbian-config --api module_webmin help` 
`armbian-config --api generate_json_options`

Tearran, 2024-11-23:
To dynamically  change the menu sub sub menu titles  `update_sub_submenu_data` in `config.runtime.sh` 

Adding a Data point should not break things. but will need to refactor the doc and generate_json_options. 
```
["module_name,port"]="<port number>"
```
Updating the menu to show the port 

```
update_sub_submenu_data "ID" "SUB_ID" "SUB_SUB_ID" "$LOCALIPADD:${module_options["module_name,port"]}"
```
SImple verify 
example  cockpit and webmin
```
# Sub sub menu updates
cockpit_port=$(lsof -iTCP:9090 -sTCP:LISTEN | grep systemd )
if [[ -n "$cockpit_port" ]]; then
	update_sub_submenu_data "Software" "Management" "MAN004" "https://$LOCALIPADD:9090"
fi
#
webmin_port=$(lsof -iTCP:10000 -sTCP:LISTEN | grep webmin )
if [[ -n "$webmin_port" ]]; then
	update_sub_submenu_data "Software" "Management" "MAN005" "https://$LOCALIPADD:10000"
fi
```
containers would be something like `| grep docker-pr `

Tearran, 2024-11-24:
### Existing keys 

- [x] id 
- [x] sub  
- [x] description
- [x] about
- [x] command
- [x] status
- [x] condition
- [x] example

# Suggested keys
- [ ] port
- [ ] service
- [ ] Reference
- [ ] maintainer 
- [ ] contributor
- [ ] parent_id