armbian/configng #267: apt-get claims missing PUBLIC KEY for "configng stable InRelease"

sauerbi, 2024-11-22:
### What happened?

When calling  "`apt-get update`" the following message comes up:

> Err:5 https://github.armbian.com/configng stable InRelease
>   The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 93D6889F9F0E78D5

### How to reproduce?

after running `apt-get upgrade` to install latest version of armbian-config (ng) the NEXT call to `apt-get update` will produce the above error.

I am running 'debian bullseye' on a rock-3a system.
uname -a gives
  Linux nginx 6.1.11-rk35xx #23.02.2 SMP PREEMPT Sat Feb 18 00:05:58 UTC 2023 aarch64 GNU/Linux

### On which host OS are you running the build script and observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-11-23:
https://forum.armbian.com/topic/47137-apt-update-fails-no_pubkey-93d6889f9f0e78d5/#comment-206173

igorpecovnik, 2024-11-23:
Bullseye is unsupported as costs of maintaining it went out of our range. If someone comes out with a fix that doesn't require complete rework of packaging, which is one of the possible solutions, we will implement it, PR will be accepted. Same goes for workaround, which was proposed in other ticket related to the same problem: https://github.com/armbian/configng/issues/265

igorpecovnik, 2024-11-28:
Found a fix. Now it should work:
https://github.com/armbian/scripts/commit/9fe944b9c275116a1811f24c76714846ced39f50

closed 2024-11-28 by igorpecovnik

dimitry-ishenko, 2024-11-28:
If I may suggest, the keyring should be packaged in a Debian package. Both Debian and Ubuntu have theirs in `debian-archive-keyring` and `ubuntu-keyring` packages respectively. This has at least 2 advantages:

1. You can install the keyring like a normal package. Eg, `sudo apt install armbian-keyring`.
2. In the future, if you need to add and/or revoke a key, you can release updated package and will be propagated to the users next time they update their system.

Just my 2 cents.

igorpecovnik, 2024-12-13:
@sauerbi 

Adding key old way should work. Implementing it to the build framework.
https://github.com/armbian/build/pull/7524