armbian/configng #283: "Network > WiFi" Section do not work

chemrgineer, 2024-11-29:
### What happened?

I installed fresh image on Radxa Zero (community support). I updated with apt, and my armbian-config version: 25.2.0-trunk.103.1129.130128. I'm using my SBC from usb ethernet adaptor. After first setup, I updated all packages and tried to activate wifi connection. But armbian-config retrieves (or just says) some packages and return back to network first menu. I could not connect wifi. 

### How to reproduce?

sudo armbian-config > Network > NE001 > NE002 > wlan0: unassigned[wifi]

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-11-29:
> sudo armbian-config > Network > NE001 > NE002 > wlan0: unassigned[wifi]

Do you get list of stations around if you execute:

sudo iw $(sudo iw dev | awk '$1=="Interface"{print $2}') scan

chemrgineer, 2024-11-29:
I get list of stations and detailed properties of them. I get full log of system and uploaded [here](https://paste.next.armbian.com/ohazapuzen).

igorpecovnik, 2024-11-29:
Thanks, then this is a problem. I don't have this board, but I will try to find something similar to test. Monday ...

schwar3kat, 2024-11-30:
This is possibly the similar to one of the issues that I am experiencing.  If so, it is probably not only that board.
WiFi can be set up manually and it works on all tests.

I tried to set up WiFi from armbian-config on:
Rock 5c - Armbian_24.11.1_Rock-5c_bookworm_vendor_6.1.75_minimal
Rock 5c - Armbian_24.11.1_Rock-5c_noble_vendor_6.1.75_gnome-kisak_desktop
Orangepizeroplus - Armbian-unofficial_24.11.0-trunk_Orangepizeroplus_bookworm_current_6.6.62 (fresh unmodified build)

I've attached a pdf detailing the process to replicate the issue.

**On Rock 5c:**
"sudo iw dev | awk '$1=="Interface"{print $2}'" produces:
wlan0

"sudo iw $(sudo iw dev | awk '$1=="Interface"{print $2}') scan" **returns data from multiple APs**  

**On Orangepizeroplus**  (this may be a different issue possibly caused by 2 interfaces)
"sudo iw dev | awk '$1=="Interface"{print $2}'" produces:
wlan1
wlan0

"sudo iw $(sudo iw dev | awk '$1=="Interface"{print $2}') scan" **is not a valid command. (returns usage options)**  See attached cmd-output.txt.

"sudo iw wlan0 scan" produces expected output for multiple ap's on both boards.
"sudo iw wlan0 scan | grep 'SSID:'" lists 15 SSIDs.

[armbian-config WiFi setup issue.pdf](https://github.com/user-attachments/files/17963541/armbian-config.WiFi.setup.issue.pdf)
[cmd-output.txt](https://github.com/user-attachments/files/17963568/cmd-output.txt)

igorpecovnik, 2024-11-30:
I tried on M4 which have similar chip-set. I have added repetition and lets see if this helps.

apt update; apt upgrade

and try again.

chemrgineer, 2024-12-01:
armbian-config version: 25.2.0-trunk.107.1130.212127, nothing changed. still can not set wifi settings with armbian-config.

schwar3kat, 2024-12-01:
> What about if this has something to do with locales?
I can try with a different locale on the device.  Not sure about the router though.

schwar3kat, 2024-12-01:
@igorpecovnik 
I have made progress understanding your network_config function.
I will debug some more when I get home from work tonight.

schwar3kat, 2024-12-02:
I think that one problem is that the statement can't handle multiple word SSIDs or blank SSIDs.
The following is probably bad bash, but it goes some way toward producing a menu list.
`LIST=($(iw dev ${adapter} scan 2> /dev/null | grep -i 'frequency\|ssid\|^bss' | sed 's/\t//g' | sed 's/BSS/BSS../g' | sed '/*/d' | cut -b7-100 | sed "s/(.*//g" | xargs -n2 -d'\n' | awk '{print "\"" $2"\"","\""$1"\""}'))`

The other problem that I don't understand is that I almost always get a blank list when running it inside the function.

The attached text file is the output of iw dev wlan0 scan at my place.
[scan.txt](https://github.com/user-attachments/files/17971938/scan.txt)

schwar3kat, 2024-12-02:
Ohe result of my modified code above:
[output_from_code_above.txt](https://github.com/user-attachments/files/17973163/output_from_code_above.txt)

schwar3kat, 2024-12-02:
The result of my amateur attempt at code above.
[output_from_code_above.txt](https://github.com/user-attachments/files/17973561/output_from_code_above.txt)

chemrgineer, 2024-12-02:
I make connection by using nmcli. armbian-config bug in my system is not solved. But I solved my problem.

schwar3kat, 2024-12-03:
@igorpecovnik   I'm posting my test results in the hope that it might be helpful in solving this issue.
The issue is definitely the creation of the ap list created by:
`LIST=($(iw dev ${adapter} scan 2> /dev/null | grep 'SSID\|^BSS' | cut -d" " -f2 | sed "s/(.*//g" | xargs -n2 -d'\n' | awk '{print $2,$1}'))`

**This mostly produces a blank LIST, but sometimes produces a bad LIST that causes the script to not work.** [bad_wifi_aps.txt](https://github.com/user-attachments/files/17999380/bad_wifi_aps.txt)

I couldn't figure out a way to generate a suitable list using iw (my bash fu is weak).
I could get a usable ordered list using wpa_cli, but I can't get wpa_cli to work in the function, 
so I couldn't use this method as a replacement for a proof of concept.

**My successful test using a pre-populated file:**
If I modify armbian-config, network_config function to use `LIST=($(cat /tmp/wifi_aps.txt))` 
Then before running armbian-config, I pre-populate the file by running : 
`wpa_cli -i wlan0 scan` 
`wpa_cli -i wlan0 all_bss | sort -t$'\t' -k3 -nr | tr ' ' '_' | cut -f1,5 | awk '{print $2 " " $1}' | sed "s/^ /none /g" | sed '1d' > /tmp/wifi_aps.txt`  [wifi_aps.txt](https://github.com/user-attachments/files/17999530/wifi_aps.txt)

I get a usable menu, and I can continue to set up the sta access point.

schwar3kat, 2024-12-04:
Blank lists seem to be a timing issue:
If I insert `sleep 4` before LIST=() then the list gets populated.

This POC fix works for my location.  
I don't know enough about WiFi and scripting to have confidence that this will work for all.
It lists ap's in order of signal strength.
```
sleep 4
LIST=()
LIST=($(iw dev ${adapter} scan 2> /dev/null | grep -i 'ssid\|^bss\|signal' | sed 's/\t//g' | sed 's/: /~/g' | sed 's/BSS /BSS~/g'| cut -d '~' -f 2 | sed '/*/d' | sed "s/(.*)//g" | pr -3 -a -t -s | sort -t$'\t' -k2 -nr | tr ' ' '_' | cut -f1,3 | awk '{print $2 " " $1}' | sed "s/^ /none /g"))
```

igorpecovnik, 2024-12-04:
> seem to be a timing issue

This is what I thougt.

> Currently it lists SSIDs with spaces with underscores

I think I have a solution for this. Trying and will implement both fixes if it works.

schwar3kat, 2024-12-05:
**On Orangepizeroplus, I could not break it for simple use.** 
I tested SSIDs with spaces, multiple SSIDs on the same adapter, and SSIDs on multiple adapters.

**On Rock 5 there is an issue with the dialog showing the setup.**  It hangs intermittently about  2/3 of attempts.   
Symptoms:  Dialog with setup shows with `<Ok>` not highlighted red.  Clicking and enter do nothing. Escape works.
I'm guessing that on this fast board, there might be another type of timing issue.  
This is not completely unworkable, it just takes a few attempts to get it working.
[image]

I will test some more on the weekend.

igorpecovnik, 2024-12-05:
> On Orangepizeroplus, I could not break it for simple use.

OK, then lets merge this as current solution is quite unstable.

> On Rock 5 there is an issue with the dialog showing the setup.

This part needs investigation. Overall, Network management have a room for improvement.

schwar3kat, 2024-12-05:
Yes, I agree.   

My perception of the issue is hazy, and I can't completely rule out user error yet.
I will try to debug a bit more on the weekend.   Time is scarce until then.

igorpecovnik, 2024-12-05:
> I will try to debug a bit more on the weekend. Time is scarce until then.

Understand. One problem at a time. I will also try to add some clarity to this networking section and split functions apart as current way is not the best. This part was done early in development of this tool, before design / coding was fixed.

I found out that there are some issues with Whiptail itself and here we have little options, but I think it is possible to relatively easy (there are few exceptions that needs to be addressed) to switch to Dialog instead. @Tearran designed this in a way that we can switch.

closed 2024-12-05 by igorpecovnik

schwar3kat, 2024-12-07:
I did a little debugging.
`show_message <<< "$(netplan get all)"` does not work with whiptail.

I replaced `show_message <<< "$(netplan get all)"` with:
`net_all="$(netplan get all)"`
`$DIALOG --title "$TITLE" --msgbox "$net_all" 0 0`

This works with whiptail.