armbian/configng #288: Changing keyboard layout doesn't work

Qwertylala1, 2024-12-01:
### What happened?

When I select the option "LO003   -  Change Keyboard layout " I'm not given any choices, instead update-initramfs runs immediately and armbian-config exits.

I'm running a server/CLI image of Armbian Bookworm with the latest updates installed on an Orange Pi 5 Plus. Armbian-config was updated today to 24.11.2 but it did the same thing with the previous version (24.11.1.1111.090121).

### How to reproduce?

Choose LO003   -  Change Keyboard layout option in armbian-config.

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-12-01:
Confirmed. Looking into.

igorpecovnik, 2024-12-01:
There is nothing extraordinary behind this command. It just calls:

`dpkg-reconfigure keyboard-configuration` 

which works fine on Odroid M1 running same Bookworm OS as a base. I could reproduce this on some old Orangepizeroplus ... which doesn't have a console ... 

I would say this problem is not within this utility but its something related to the kernel. Lets' keep this issue open if someone got some ideas ...

leggewie, 2025-04-05:
So, the question is, what happens when @Qwertylala1 runs that command `dpkg-reconfigure keyboard-configuration`. Can you tell us, please?

FWIW, I was also not given anything to choose from on a Noble system on BPi M2+ Allwinner H3 SOC.

Tearran, 2025-04-05:
@leggewie We aim to be transparent — the code is open, and the behavior of armbian-config is documented across the web, the repository, and within markdown and HTML resources. We even include a built-in manual. If that doesn’t meet your expectations, perhaps it’s not a lack of transparency but a mismatch in how you’re looking for information. We’re open to specific suggestions, but blanket statements like “it’s a black box” don’t reflect the effort already made to demystify the tool.