armbian/configng #291: Support ticket for 3rd party distributions

igorpecovnik, 2024-12-02:
### What happened?

a

### How to reproduce?

a

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2024-12-02:
"Manual install over Armbian" @way5 what is the reason for migration?

way5, 2024-12-02:
The reason I'd like to stay with DietPi is that quite a few of the scripts I am using are normally live in its environment, so I didn't plan to rewrite anything hoping things would stay as is. 
However if the DietPi install is not possible over Armbian, I'll do the migration.

Thanks anyways!

igorpecovnik, 2024-12-02:
> quite a few of the scripts I am using are normally live in its environment

Can you provide a list of those scripts? Adding scripts is super easy job vs. kernel maintenance what dietpi doesn't do at all.

igorpecovnik, 2024-12-02:
> Since Dietpi is actually based on Armbian

It can't be otherwise. The problem is their complete absence of contribution to common problems, paying common costs. In 10 years firs person made < 5, and other two not a single contribution to any open source project.

Dietpi is a lot of noise (marketing) while in reality is a spaghetti install / config script made by one person. Everything else, true value, biggest pain, comes from kernel maintainers around Armbian and other similar projects https://docs.armbian.com/#comparison 

> while preparing the next Diet-build

I think you already prepare more value around this hardware then they ever will. Providing right information and test. This is [what we expect from maintainers for particular board](https://forum.armbian.com/staffapplications/application/8-single-board-computer-maintainer/) to do 4 times per year. 

> I also must admit that the unofficial build which was shared with me by @CodeChenL works flawlessly and much smoother

Great! I hope he made a PR to the main code as this code maintainace can only be common work -  we have absolutely no ways to pick and integrate solutions people made and post to forums around. Its estimated that we would need several full time developers to do that. Even this is heavily in community interest, there is nothing we can do. Also because project such as Dietpi exists. Everything we invest in common, they also get, while they invests noting.

way5, 2024-12-02:
Alright. Thank you for the info. Now I am gonna keep up with Armbian. I have already did the checkup in the past few days for the DTSs and the pins seem to be mapped fine now. It was a pain to access hardware in previous versions so a couple of years I had kernel updates disabled.

I appreciate what you're doing guys. Hope I could help you in some ways in future.

Cheers

CodeChenL, 2024-12-03:
Hi @igorpecovnik,
My main concern about not submitting a PR to Armbian at the moment is that these patches seem a bit hacky. However, to ensure compatibility with eMMC chips from different manufacturers (mainly Samsung), this is currently the only solution we (Radxa) have been able to find. These patches have been validated by several customers and ourselves. I'll submit a pr first, to make it easier for us to discuss how to deal with it, and to not stray from the topic of this issues.

igorpecovnik, 2024-12-03:
> My main concern about not submitting a PR to Armbian at the moment is that these patches seem a bit hacky

Understand. TBH we have several such patches in the system. More then absence of a (proper) fix is having no fix / workaround at all. In most cases the one that submitted the patch, improves this within time. Sometimes, if we really to the proper solution, patches are sent upstream (where acceptance standards are rigid - for a reason).

>  I'll submit a pr first,

That is probably best. Let other people, that are close, comment / suggests changes.

way5, 2024-12-03:
Right after the image has been written to EMMC the kernel got started fine. However once I install the bootloader (`armbian-config -> SY006`) onto `/dev/mmcblk1` it stops booting again. The bootloader messages are exactly the same as [here](https://forum.radxa.com/t/rzero-autoboot-from-emmc/24764)

igorpecovnik, 2025-01-25:
> Okay, thanks for testing on your side, if I understand correctly there's just waiting for an update of the rockchip kernel? Hoping it will be solved

@Seb-tech77
Dietpi download this and all other valuable fixes from [us](https://docs.armbian.com/#what-is-armbian). then they sell it to you as they would do it. Follow release logs https://docs.armbian.com/Release_Changelog/ if you want to find out when this is fixed. This is never fixed by Rockchip, but people that contribute here https://github.com/armbian/linux-rockchip 

This problem is well known for awhile and it won't be solved just like that. Can take weeks to months.

Solution that works today (with some other problems) is https://www.armbian.com/orangepi-5/ K6.12 and official OMV install. Armbian have no bloatware https://docs.armbian.com/#key-advantages