armbian/configng #311: SOS

dimitry-ishenko, 2024-12-11:
### What happened?

s

### How to reproduce?

s

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Tearran, 2024-12-11:
@dimitry-ishenko  I did something like that on the other config project once. IMO a happy accident, was nice to see it bar for a mi lol. Many of them do not need restored. As outdated or examples to concept.  

```
options_to_json
gui-mouse
Tearran-patch-1
259-task-replace-existing-keys-with-a-single-consistent-key-list
```

dimitry-ishenko, 2024-12-11:
Thanks @Tearran. I've managed to restore them all. 😓 Feel free to nuke them if they are not needed. Sorry for all the noise. It's been a long day and then I decided to do some coding before going to bed... 🤯

Tearran, 2024-12-11:
ooh, that would be good for https://github.com/armbian/configng/wiki

Think it would be nice to have some git-fu bits like that.

igorpecovnik, 2024-12-11:
> It's been a long day and then I decided to do some coding before going to bed..

No worries, glad things were restored quickly and we are back in the game. And we learned something new.

closed 2024-12-11 by igorpecovnik