armbian/configng #319: [Bug]: Failed to fetch github.armbian.com

way5, 2024-12-12:
### What happened?

Hey everyone! What is happening with github.armbian.com? it is empty now:

```bash

    _             _    _                             __  __ _    _      _ 
   /_\  _ _ _ __ | |__(_)__ _ _ _ ___ _  _ _ _  ___ / _|/ _(_)__(_)__ _| |
  / _ \| '_| '  \| '_ \ / _` | ' \___| || | ' \/ _ \  _|  _| / _| / _` | |
 /_/ \_\_| |_|_|_|_.__/_\__,_|_||_|   \_,_|_||_\___/_| |_| |_\__|_\__,_|_|
                                                                          
 v24.11 rolling for Radxa Zero running Armbian Linux 6.6.65-current-meson64

 Packages:     Debian stable (bookworm)
 Updates:      Kernel upgrade enabled and 7 packages available for upgrade 
 Support:      DIY (custom image)
 IP addresses: (LAN) IPv4: 10.0.0.1 IPv6: fe80::2250:e7ff:fed2:63a6 (WAN) 187.191.8.243

 Performance:  

 Load:         19%           	Up time:       1 min	
 Memory usage: 3% of 3.68G  	
 CPU temp:     44°C           	Usage of /:    4% of 57G    	

 Commands: 

 Configuration : armbian-config
 Upgrade       : armbian-upgrade
 Monitoring    : htop

Last login: Wed Dec 11 18:00:59 2024
root@radxa-zero:~# apt update
Hit:1 http://security.debian.org bookworm-security InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease                                    
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease                            
Hit:4 http://deb.debian.org/debian bookworm-backports InRelease                          
Hit:6 https://github.armbian.com/configng stable InRelease                  
Hit:5 http://fi.mirror.armbian.de/apt bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
7 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@radxa-zero:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  armbian-bsp-cli-radxa-zero-current armbian-config armbian-firmware base-files
  linux-dtb-current-meson64 linux-image-current-meson64 linux-u-boot-radxa-zero-current
7 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 111 kB/137 MB of archives.
After this operation, 743 kB disk space will be freed.
Do you want to continue? [Y/n] 
Err:1 https://github.armbian.com/configng stable/main arm64 armbian-config all 20241013-1817
  404  Not Found [IP: 185.199.108.153 443]
E: Failed to fetch https://github.armbian.com/configng/pool/main/a/armbian-config/armbian-config_20241013-1817_all.deb  404  Not Found [IP: 185.199.108.153 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@radxa-zero:~# apt --fix-missing install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
root@radxa-zero:~# 

```

### How to reproduce?

```bash

root@radxa-zero:~# apt update
root@radxa-zero:~# apt upgrade

```

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-12-12:
Duplicate of https://github.com/armbian/configng/issues/316

closed 2024-12-12 by EvilOlaf

igorpecovnik, 2024-12-12:
There is indeed something wrong. Looking into ...

igorpecovnik, 2024-12-12:
Fixed. Had to remove and recreate repository.