armbian/configng #322: Question about get_user_continue

dimitry-ishenko, 2024-12-14:
### Task description

@Tearran trying to understand what's going on inside `get_user_continue` function. In the code it's always used with `process_input`. And if I select No when asked "Do you wish to continue?" the program just exits instead of returning to the menu...

Why doesn't `get_user_continue` just return 0 or 1 codes? Or is there a deeper design plot that I've failed to grasp?

Tearran, 2024-12-14:
>Why doesn't get_user_continue just return 0 or 1 codes? Or is there a deeper design plot that I've failed to grasp?
Simply put, return breaks the group. Saying no continus and installs anyways. 
The reason  this exist is to offer a warning and/or user agreement to continue. before system files are changed. Can use some refinement. Even refactored. 

This probably over complex setup was to be used for inclusion for the nuanced difference of whiptail and dialog and read.

Background I am not a fan of loops or nesting in bash haven't had need for returns when piping echos. The UX aspects aside, The approach I prefer to take with bash is do linear procedure, pipe out or exit.   

 
> And if I select No when asked "Do you wish to continue?" the program just exits instead of returning to the menu...

This has been a point before, The reason why return was not looked into. responding `no` to `do you want to continue` does not translates to continuing the loop.  

>Note: my selfish reason so i don't have to going back, back and back and sometime back again to be able to exit is cumbersome.   `CTL-C` needs to be implemented or fixed.

dimitry-ishenko, 2024-12-14:
I see, basically the reason I asked is that my idea of closing stdin for non-interactive mode (https://github.com/armbian/configng/pull/320#issuecomment-2541973414) has one gotcha.

If the function being called with `--api` tries to show a dialog asking for user input (eg, the one shown by `get_user_continue`) the program will hang (since stdin is closed and there is no way for the user to input anything).

In non-interactive mode (with `--api`) we don't want user interaction anyway, so I was thinking of having this function do a similar check for stdin (`[[ -t 0 ]]`) and if it's closed, assume the answer is yes.

What do you think?

Tearran, 2024-12-15:
Thanks to your examples the concept as I understand is quite good and seems the right directions, would help with keeping the UX/UI completely separated from the modules.

>NOTE The '--api' is useful for troubleshooting and testing with UX and NonUX, as it is. And a goal or refining to more of an api of sorts. 

That and your suggestions with the error handling and sanitize Perhaps we need to split the `--api`  option again as we did with '--cmd', Might be a good place for some clarity to make a case '--noint'

closed 2025-03-26 by dimitry-ishenko