armbian/configng #324: [Bug]: `sanitize_input` does not work as expected

dimitry-ishenko, 2024-12-14:
### What happened?

While testing PR #320 I would get the following error:
```shell
admin@armbian:~/configng$ bin/armbian-config --api pkg_install vim-airline

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package Invalid argument: vim-airline
```

Upon further testing, any package containing dash (`-`) in the name would produce similar error.

It turns out the error is generated by the `sanitize_input` function which does not allow dashes. However, the pipeline doesn't seem to work as expected, since instead of just reporting the error "Invalid argument: vim-airline," this output is passed further down the pipeline.

In the end the `pkg_install` function tries to install a package called `Invalid argument: vim-airline`, which results in a confusing error.

### How to reproduce?

See above.

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dimitry-ishenko, 2024-12-14:
Looking at

https://github.com/armbian/configng/blob/0a257fda5e14b5a5719478568a426cf4b27fa405/tools/modules/functions/config_interface.sh#L480-L491

My guess is that the intended behavior was to terminate program execution and print out an error message. However, in most cases `sanitize_input` will be used in a sub-shell such as

https://github.com/armbian/configng/blob/0a257fda5e14b5a5719478568a426cf4b27fa405/bin/armbian-config#L94
or
https://github.com/armbian/configng/blob/0a257fda5e14b5a5719478568a426cf4b27fa405/bin/armbian-config#L106

This will terminate the sub-shell with exit code 1, but unless checked in the super-shell, will continue normal execution, which is what's happening here.

dimitry-ishenko, 2024-12-14:
Suggestions to resolve the bug:

- When reporting an error, direct it at stderr and not stdout (which is why we have it 😉 ), eg:
  ```
  echo "Invalid argument: $arg" >&2
  exit 1
  ```
- Check for sub-shell exit code:
  ```
  args=$(sanitize_input "$@") || exit 1
  ```
- We can also leverage `set -e` option, which will automatically abort script execution on error.

- To add consistency and reduce boiler-plate we could introduce commands such as:
  ```
  error() { echo "$@" >&2; }
  die() { error "$@"; exit 1; }
  ```
  Then the above code becomes something like:
  ```bash
  sanitize_input() {
      ...
          die "Invalid argument: $arg"
      ...
  }
  ```
  and
  ```bash
  args=$(sanitize_input "$@") || die
  ```

dimitry-ishenko, 2024-12-14:
And, obviously, `sanitize_input` should allow dashes in the input.

Tearran, 2024-12-15:
@dimitry-ishenko Sanitise input contribution was aimed primarily at removing escape sequences. 
This is one of the placeholder implementations that is incomplete and needs refinement or replacement. Another example you are familiar with is `process_input`.

This has been a place for expanding and learning: If a may ask :) Are you saying that while the implementation is flawed (BUG), the concept is not, and `sanitize_input` should handle input validation more comprehensively rather than be removed? 

For later implementation and though. 
If we validate, should we not include a manner to use the database to actually verify inputs?

dimitry-ishenko, 2024-12-15:
> Are you saying that while the implementation is flawed (BUG), the concept is not, and `sanitize_input` should handle input validation more comprehensively rather than be removed?

It's hard to tell without knowing the original intent, but I would say it should be removed... I am not a security specialist, but I don't see what problem it's trying to solve... Is there a case of using an escape sequence to compromise the system?

Or, if there is a valid need for it, then maybe it can be modified to use a blacklist to detect escape char in the input instead.

Tearran, 2024-12-15:
The intent was precaution more than anything for when armbian-config was not strictly an admin tool as it is now.

dimitry-ishenko, 2024-12-15:
Maybe it should be removed for now, and if the need arises later, it can be resurrected. Just my 2 cents.

dimitry-ishenko, 2025-01-29:
@Tearran if you don't mind, I'll do a small PR that drops `sanitize_input` like you did in #326?

I want to get it, so I can finish my service wrapper PR (#323) and then Samba. Thanks.

PR #408 (2025-02-04): Fix sanitize
# Description

Fixes `sanitize_input` command used with the `--api` and `--cmd` options.

Issue reference:  Closes #324.

# Implementation Details

_Provide a detailed description of the implementation. Include the following:_

- [x] Key changes introduced by this PR

Replaced `sanitize_input` function with simplified version called `sanitize`. Sanitation is only done on the first argument (which is a function name in case of `--api`, or a command name in case of `--cmd`).

_UPDATE:_ The final commit fixes some flawed tests, whose failure was masked by explicit `exit 0` statements [... cut]
closed 2025-02-04 by igorpecovnik