armbian/configng #337: [Bug]: kernel switch failed

EvilOlaf, 2024-12-24:
### What happened?

Tried to switch from `current` to `edge` kernel on OPi1+ (sunxi64 to say).
Did not survive reboot. Mounted the sdcard and for unknown reason the symlinks for /boot/Image and uInitrd were dead end.
Used `dpkg` to install an image and dtb package via `chroot` to fix it.

### How to reproduce?

Did not try to reproduce nor did any further investigation. Since that was my local dns my primary goal was to get it back online asap :grin

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2024-12-24:
Go home, you're drunk lol

igorpecovnik, 2024-12-24:
This was the problem in old builds ... Does this happens on fully upgraded armbian-config?

EvilOlaf, 2024-12-24:
I freshly installed armbian-config before attempting since it wasn't installed.

igorpecovnik, 2024-12-24:
When you install image, first upgrade of armbian-config adds repository, second is updating from this. Just make sure ... will add build date to the welcome screen soon.

EvilOlaf, 2024-12-24:
Yeah I see. First upgrade indeed bumped to 25.2 trunk

igorpecovnik, 2024-12-24:
https://github.com/armbian/configng/commit/023bc8dd291c42342cb567e9a3d1747fc502b308

EvilOlaf, 2024-12-26:
Assuming it was fixed already and kernel switch failed due to outdated armbian-config package

closed 2024-12-26 by EvilOlaf