armbian/configng #338: [Feature Request]: Update issue template

EvilOlaf, 2024-12-24:
### Which feature would you like to have?

The issue template asks for the host OS which is probably a c&p mistake from `build` repo's template.
Two options:
- remove this block entirely
- rephrase to ask for Armbian OS flavor and version.

Didn't send a PR already since I'm not sure which is preferred.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

Tearran, 2024-12-25:
>Didn't send a PR already since I'm not sure which is preferred.

The unit test indicates, Stable Debian, Ubuntu jammy and noble
***
>rephrase to ask for Armbian OS flavor and version

perhaps Something like 

On which flavor of Armbian are you running the armbian-config script and observing this problem?

- [ ] Armbian Debian Bookworm
- [ ] Armbian Ubuntu Jammy
- [ ] Armbian Ubuntu Noble

If it is not one of the above, it may not be an Armbian bug. Consider: Making a [Feature request](https://github.com/armbian/configng/issues/new?assignees=&labels=Feature+Request&projects=&template=feature-reqests.yml&title=%5BFeature+Request%5D%3A+) and  [Sponsor Armbian](https://github.com/sponsors/armbian) ...

Tearran, 2024-12-29:
>The issue template asks for the host OS which is probably a c&p mistake from build repo's template.
Two options:

IMO If the block does not trigger events, remove it :)

closed 2025-01-23 by igorpecovnik