armbian/configng #343: [Bug]: No overlay config shown

EvilOlaf, 2024-12-27:
### What happened?

Station P1 Armbian 25.02.0-trunk user-built 6.13.0-rc4-edge-rockchip64 (Debian bookworm)
armbian-config version 25.2.0-trunk.229.1227.143514 (upgraded right before opening this report)
https://paste.next.armbian.com/qemubiluce

When navigating to `System - Kernel` the option for overlays is missing. Only SY201 and SY204 are shown.
Though overlays are available:

```bash
root@station-p1:/boot/dtb/rockchip/overlay# ls|grep rk3399
rockchip-rk3399-dwc3-0-host.dtbo
rockchip-rk3399-i2c7.dtbo
rockchip-rk3399-i2c8.dtbo
rockchip-rk3399-opp-2ghz.dtbo
rockchip-rk3399-pcie-gen2.dtbo
rockchip-rk3399-spi-jedec-nor.dtbo
rockchip-rk3399-spi-spidev.dtbo
rockchip-rk3399-uart4.dtbo
rockchip-rk3399-w1-gpio.dtbo
```

This seems to affect other boards too. Just right now a person in Discord reported this with Jetson Nano

### How to reproduce?

Get a Station P1 or Jetson Nano and try enabling overlays via armbian-config

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

aperture147, 2024-12-27:
Confirmed. Happening to me with Armbian 23.8.1 Bookworm minimal on Jetson Nano

Tearran, 2024-12-28:
Works on `meson64`
 `./bin/armbian-config --cmd SY210`
`./bin/armbian-config --api manage_dtoverlays`

```
grep 'LINUXFAMILY' /etc/armbian-release | sed 's/^LINUXFAMILY=//g
```
`meson64`

Dose the command 
```bash
grep 'LINUXFAMILY' /etc/armbian-release | sed 's/^LINUXFAMILY=//g'
```
match the any of the related` "$LINUXFAMILY" == "*"` `https://github.com/armbian/configng/blob/fb0d08a50f99dc856810648d789fdbfde30fb6f3/tools/modules/system/manage_dtoverlays.sh#L20

EvilOlaf, 2024-12-28:
Yes, matches `rockchip64`.
Might the folder name be the culprit? It is named `rockchip`only.

Tearran, 2024-12-28:
>"obviously fails to apply changes since Station P1 uses extlinux instead of plain armbianEnv.txt"

Not so obvious, thank you for sharing it was helpful 💯. Looking further, we should add overlay support for extlinux.conf.

>"When navigating to System - Kernel the option for overlays is missing. Only SY201 and SY204 are shown."

This is the expected behavior in this case, will only show if compatible, as we currently only have armbianEnv.txt compatibility. Should not show.

[Reference: config.system.json](https://github.com/armbian/configng/blob/fb0d08a50f99dc856810648d789fdbfde30fb6f3/tools/json/config.system.json#L50)

@EvilOlaf, in the context of code, this is a "[feature request]"—an oversight perhaps :), but it is still a "[feature request]" nevertheless. 

>"Calling directly with armbian-config --cmd SY210 brings up the menu correctly."

I can see perhaps `brings up the menu correctly` being a bug and will continue that on a similar issue #344.

EvilOlaf, 2024-12-29:
> Looking further, we should add overlay support for extlinux.conf.

Not sure if its worth the effort since barely any images are using extlinux. I think those who are are all csc?

> This is the expected behavior in this case, will only show if compatible, as we currently only have armbianEnv.txt compatibility. Should not show.

Ah okay, Didn't know the show-up depends basically on armbianEnv being there. 

>  in the context of code, this is a "[feature request]"—an oversight perhaps :),

Yes. The question now is to either mangle this issue report or simply create a new one requesting support for extlinux ;). Your call.

Tearran, 2024-12-29:
>Not sure if its worth the effort since barely any images are using extlinux. I think those who are are all csc?

Agreed closing.

closed 2024-12-29 by Tearran