armbian/configng #344: No overlays are shown

MMinga, 2024-12-27:
### What happened?

There is now possibilty to change settings when nothing is shown.
[image]
[image]

### How to reproduce?

no description needed it never has shown devices.

### On which host OS are you running the build script and observing this problem?

Ubuntu 22.04 Jammy

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Tearran, 2024-12-28:
works on `meson`
```
armbian-config --api manage_dtoverlays
```
[image]

igorpecovnik, 2025-01-23:
This should be fixed. I have tried with latest Bookworm image + latest armbian-config (on Odroid M1). 

[image]

Try update everything and if it still doesn't work, provide info from:

```
sudo apt -y install tree
tree /boot/dtb/
cat /boot/armbianEnv.txt
```

sp2ong, 2025-01-24:
**On Orange Pi Zero V1** after update & upgrade when selecting Manage oerlays from armbian-config not see my selected modules in overlays= in armbianEnv.txt so I selected again but still added overlay_prefix to modules in overlays= and after reboot no modules loaded. When removing manually overlay prefix form names modules in overlaus= all works
OK

_v25.2 rolling for Orange Pi Zero running Armbian Linux 6.6.72-current-sunxi_

_cat /boot/armbianEnv.txt 
verbosity=1
bootlogo=false
console=both
disp_mode=1920x1080p60
overlay_prefix=sun8i-h3
overlays=sun8i-h3-analog-codec sun8i-h3-i2c0 sun8i-h3-uart1 sun8i-h3-uart2 sun8i-h3-usbhost2 sun8i-h3-usbhost3
rootdev=UUID=79665e18-ad8c-4276-a34d-3b181f8f6fac
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u_

tree /boot/dtb/
/boot/dtb/
├── overlay
│   ├── README.sun4i-a10-overlays
│   ├── README.sun5i-a13-overlays
│   ├── README.sun7i-a20-overlays
│   ├── README.sun8i-h3-overlays
│   ├── sun4i-a10-analog-codec.dtbo
│   ├── sun4i-a10-can.dtbo
│   ├── sun4i-a10-fixup.scr
│   ├── sun4i-a10-i2c1.dtbo
│   ├── sun4i-a10-i2c2.dtbo
│   ├── sun4i-a10-nand.dtbo
│   ├── sun4i-a10-pps-gpio.dtbo
│   ├── sun4i-a10-pwm.dtbo
│   ├── sun4i-a10-spdif-out.dtbo
│   ├── sun4i-a10-spi-jedec-nor.dtbo
│   ├── sun4i-a10-spi-spidev.dtbo
│   ├── sun4i-a10-uart2.dtbo
│   ├── sun4i-a10-uart3.dtbo
│   ├── sun4i-a10-uart4.dtbo
│   ├── sun4i-a10-uart5.dtbo
│   ├── sun4i-a10-uart6.dtbo
│   ├── sun4i-a10-uart7.dtbo
│   ├── sun4i-a10-w1-gpio.dtbo
│   ├── sun5i-a13-analog-codec.dtbo
│   ├── sun5i-a13-fixup.scr
│   ├── sun5i-a13-i2c1.dtbo
│   ├── sun5i-a13-i2c2.dtbo
│   ├── sun5i-a13-nand.dtbo
│   ├── sun5i-a13-pwm.dtbo
│   ├── sun5i-a13-spi0.dtbo
│   ├── sun5i-a13-spi1.dtbo
│   ├── sun5i-a13-spi2.dtbo
│   ├── sun5i-a13-spi-jedec-nor.dtbo
│   ├── sun5i-a13-spi-spidev.dtbo
│   ├── sun5i-a13-uart0.dtbo
│   ├── sun5i-a13-uart1.dtbo
│   ├── sun5i-a13-uart2.dtbo
│   ├── sun5i-a13-uart3.dtbo
│   ├── sun7i-a20-analog-codec.dtbo
│   ├── sun7i-a20-can.dtbo
│   ├── sun7i-a20-fixup.scr
│   ├── sun7i-a20-i2c1.dtbo
│   ├── sun7i-a20-i2c2.dtbo
│   ├── sun7i-a20-i2c3.dtbo
│   ├── sun7i-a20-i2c4.dtbo
│   ├── sun7i-a20-mmc2.dtbo
│   ├── sun7i-a20-nand.dtbo
│   ├── sun7i-a20-pps-gpio.dtbo
│   ├── sun7i-a20-pwm.dtbo
│   ├── sun7i-a20-spdif-out.dtbo
│   ├── sun7i-a20-spi-add-cs1.dtbo
│   ├── sun7i-a20-spi-jedec-nor.dtbo
[... 167 lines cut ...]
├── sun8i-h3-orangepi-pc-plus.dtb
├── sun8i-h3-orangepi-plus2e.dtb
├── sun8i-h3-orangepi-plus.dtb
├── sun8i-h3-orangepi-zero-plus2.dtb
├── sun8i-h3-rervision-dvk.dtb
├── sun8i-h3-zeropi.dtb
├── sun8i-r16-bananapi-m2m.dtb
├── sun8i-r16-nintendo-nes-classic.dtb
├── sun8i-r16-nintendo-super-nes-classic.dtb
├── sun8i-r16-parrot.dtb
├── sun8i-r40-bananapi-m2-ultra.dtb
├── sun8i-r40-oka40i-c.dtb
├── sun8i-s3-elimo-initium.dtb
├── sun8i-s3-lichee-zero-plus.dtb
├── sun8i-s3-pinecube.dtb
├── sun8i-t113s-mangopi-mq-r-t113.dtb
├── sun8i-t3-cqa3t-bv3.dtb
├── sun8i-v3-sl631-imx179.dtb
├── sun8i-v3s-licheepi-zero-dock.dtb
├── sun8i-v3s-licheepi-zero.dtb
├── sun8i-v40-bananapi-m2-berry.dtb
├── sun9i-a80-cubieboard4.dtb
└── sun9i-a80-optimus.dtb

2 directories, 241 files

sp2ong, 2025-01-24:
armbian-config was that if there is overlay_prefix= in armbianEnv.txt then the list of device modules was shown only for those that have overlay_prefix and not as currently all modules are shown for different platforms e.g. for OZPI I see suni8i-h3- sun4i-a10 etc. which is less readable. If only those that have overlay_prefix in rambianEnv.txt were shown as it used to be it is more readable but this is just a small suggestion from the user's point of view

sp2ong, 2025-01-24:
Hm

in result: tree /boot/dtb/  I see only

├── sun8i-h3-orangepi-2.dtb
├── sun8i-h3-orangepi-lite.dtb
├── sun8i-h3-orangepi-one.dtb
├── sun8i-h3-orangepi-pc.dtb
├── sun8i-h3-orangepi-pc-plus.dtb
├── sun8i-h3-orangepi-plus2e.dtb
├── sun8i-h3-orangepi-plus.dtb
├── sun8i-h3-orangepi-zero-plus2.dtb

but not exit sun8i-h3-orangepi-zero.dtb  ??? for Ornage Pi Zero v1  is used  sun8i-h3-zeropi.dtb  ???

MMinga, 2025-01-24:
No on my system it does not work:

> /boot/dtb/
├── overlay
│   ├── README.rk322x-overlays
│   ├── README.rockchip-overlays
│   ├── rk322x-bt-8723cs.dtbo
│   ├── rk322x-cpu-hs.dtbo
│   ├── rk322x-cpu-hs-lv.dtbo
│   ├── rk322x-cpu-stability.dtbo
│   ├── rk322x-ddr3-330.dtbo
│   ├── rk322x-ddr3-528.dtbo
│   ├── rk322x-ddr3-660.dtbo
│   ├── rk322x-ddr3-800.dtbo
│   ├── rk322x-emmc-ddr-ph180.dtbo
│   ├── rk322x-emmc-ddr-ph45.dtbo
│   ├── rk322x-emmc.dtbo
│   ├── rk322x-emmc-hs200.dtbo
│   ├── rk322x-emmc-pins.dtbo
│   ├── rk322x-fixup.scr
│   ├── rk322x-ir-wakeup.dtbo
│   ├── rk322x-led-conf1.dtbo
│   ├── rk322x-led-conf2.dtbo
│   ├── rk322x-led-conf3.dtbo
│   ├── rk322x-led-conf4.dtbo
│   ├── rk322x-led-conf5.dtbo
│   ├── rk322x-led-conf6.dtbo
│   ├── rk322x-led-conf7.dtbo
│   ├── rk322x-led-conf8.dtbo
│   ├── rk322x-led-conf-default.dtbo
│   ├── rk322x-nand.dtbo
│   ├── rk322x-usb-otg-peripheral.dtbo
│   ├── rk322x-wlan-alt-wiring.dtbo
│   ├── rockchip-ds1307.dtbo
│   ├── rockchip-fixup.scr
│   ├── rockchip-i2c1.dtbo
│   ├── rockchip-i2c4.dtbo
│   ├── rockchip-pwm1.dtbo
│   ├── rockchip-pwm2.dtbo
│   ├── rockchip-pwm3.dtbo
│   ├── rockchip-spi0.dtbo
│   ├── rockchip-spi2.dtbo
│   ├── rockchip-spidev0.dtbo
│   ├── rockchip-spidev2.dtbo
│   ├── rockchip-uart1.dtbo
│   ├── rockchip-uart2.dtbo
│   ├── rockchip-uart3.dtbo
│   ├── rockchip-uart4.dtbo
│   └── rockchip-w1-gpio.dtbo
├── rk3036-evb.dtb
├── rk3036-kylin.dtb
├── rk3066a-bqcurie2.dtb
├── rk3066a-marsboard.dtb
├── rk3066a-mk808.dtb
├── rk3066a-rayeager.dtb
├── rk3128-evb.dtb
├── rk3188-bqedison2qc.dtb
├── rk3188-px3-evb.dtb
├── rk3188-radxarock.dtb
├── rk3228-evb.dtb
├── rk3229-evb.dtb
├── rk3229-xms6.dtb
├── rk322x-box.dtb
├── rk3288-evb-act8846.dtb
├── rk3288-evb-rk808.dtb
├── rk3288-firefly-beta.dtb
├── rk3288-firefly.dtb
├── rk3288-firefly-reload.dtb
├── rk3288-miqi.dtb
├── rk3288-phycore-rdk.dtb
├── rk3288-popmetal.dtb
├── rk3288-r89.dtb
├── rk3288-rock2-square.dtb
├── rk3288-rock-pi-n8.dtb
├── rk3288-tinker.dtb
├── rk3288-tinker-s.dtb
├── rk3288-veyron-brain.dtb
├── rk3288-veyron-fievel.dtb
├── rk3288-veyron-jaq.dtb
├── rk3288-veyron-jerry.dtb
├── rk3288-veyron-mickey.dtb
├── rk3288-veyron-mighty.dtb
├── rk3288-veyron-minnie.dtb
├── rk3288-veyron-pinky.dtb
├── rk3288-veyron-speedy.dtb
├── rk3288-veyron-tiger.dtb
├── rk3288-vyasa.dtb
├── rk3288-xt-q8l-v10.dtb
├── rv1108-elgin-r1.dtb
├── rv1108-evb.dtb
└── rv1126-edgeble-neu2-io.dtb

1 directory, 87 files
verbosity=1
extraargs=coherent_pool=2M
bootlogo=false
overlay_prefix=rockchip,rk
overlays=es90x8q2m-dac,i2c1
rootdev=UUID=a39b8cef-c3a4-41d8-a0cf-fe08d139f982
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u

igorpecovnik, 2025-01-25:
@The-going Do you have any idea what we messed up here?

The-going, 2025-01-25:
> overlay_prefix=rockchip,rk

Incorrect prefix.
In your case, it could be:
overlay_prefix=rockchip
or
overlay_prefix=rk322x

igorpecovnik, 2025-01-25:
> Incorrect prefix.

I assume this is user error? I just want to make sure our part (build framework) and armbian-config works well?

sp2ong, 2025-01-25:
But I have on my Orange Pi Zero V1

_overlay_prefix=sun8i-h3
overlays=sun8i-h3-analog-codec sun8i-h3-i2c0 sun8i-h3-uart1 sun8i-h3-uart2 sun8i-h3-usbhost2 sun8i-h3-usbhost3_

and when is added overlay prefix in overlays modules not loaded

The-going, 2025-01-25:
> > Incorrect prefix.
> 
> I assume this is user error? I just want to make sure our part (build framework) and armbian-config works well?

We do not list the overlay prefix in any download script.
There can only be one overlay prefix.

This is user error.

The-going, 2025-01-25:
> But I have on my Orange Pi Zero V1
> 
> _overlay_prefix=sun8i-h3 overlays=sun8i-h3-analog-codec sun8i-h3-i2c0 sun8i-h3-uart1 sun8i-h3-uart2 sun8i-h3-usbhost2 sun8i-h3-usbhost3_
> 
> and when is added overlay prefix in overlays modules not loaded

**_uname -r_**

sp2ong, 2025-01-25:
uname -r
6.6.72-current-sunxi

sp2ong, 2025-01-25:
I sent my all information yesterday: https://github.com/armbian/configng/issues/344#issuecomment-2611696552

sp2ong, 2025-01-25:
I use image https://www.armbian.com/orange-pi-zero/

Minimal/IOT images with Armbian Linux v6.6

Build Date: Jan 23, 2025

which updated

The-going, 2025-01-25:
> overlays=sun8i-h3-analog-codec sun8i-h3-i2c0 sun8i-h3-uart1 sun8i-h3-uart2 sun8i-h3-usbhost2 sun8i-h3-usbhost3

Valid:
```
overlay_prefix=sun8i-h3
overlays=analog-codec i2c0 uart1 uart2 usbhost2 usbhost3
```
Name DTBO:  ${overlay_prefix}-${overlay_file}.dtbo
See: 
[load fdt](https://github.com/armbian/build/blob/fbd8c8e4c904c31d2bcbcd77ff7b240c8a2f2175/config/bootscripts/boot-sunxi.cmd#L106)

sp2ong, 2025-01-25:
Yes, when no exist overly prefix in overlays is all working but armbian-config when I manage Overlays added overlay prefix to overlays

The-going, 2025-01-25:
> when selecting Manage oerlays from armbian-config not see my selected modules in overlays= in armbianEnv.txt so I selected again but still added overlay_prefix to modules in overlays= and after reboot no modules loaded. When removing manually overlay prefix form names modules in overlaus= all works

@igorpecovnik This looks like an error in the armbian config script.

sp2ong, 2025-01-25:
I and others wrote about this at: https://github.com/armbian/configng/issues/360

sp2ong, 2025-01-25:
Similar problem with latest armbian-config NG exit on Orange Pi Zero 3 with Bookworm Armbian Image which is updated and armbian-config add overlay prefix to overlays which causes modules not to load, I must have to manually remove overly prefix in amrbianEnv.txt

igorpecovnik, 2025-01-25:
> This looks like an error in the armbian config script.

Am I blind, but I don't see any problems here. Prefix is not touched - it has to set right at build time.

https://github.com/armbian/configng/blob/main/tools/modules/system/manage_dtoverlays.sh#L48-L50

The-going, 2025-01-25:
> > This looks like an error in the armbian config script.
> 
> Am I blind, but I don't see any problems here. Prefix is not touched - it has to set right at build time.
> 
> https://github.com/armbian/configng/blob/main/tools/modules/system/manage_dtoverlays.sh#L48-L50

Okay, I'll take a look tonight.
Now I want to finish the BPI-M4-Berry.

MMinga, 2025-01-25:
> > overlay_prefix=rockchip,rk
> 
> Incorrect prefix. In your case, it could be: overlay_prefix=rockchip or overlay_prefix=rk322x

Corrected but also the overlays are not shown.

`/boot/dtb/
├── overlay
│   ├── README.rk322x-overlays
│   ├── README.rockchip-overlays
│   ├── rk322x-bt-8723cs.dtbo
│   ├── rk322x-cpu-hs.dtbo
│   ├── rk322x-cpu-hs-lv.dtbo
│   ├── rk322x-cpu-stability.dtbo
│   ├── rk322x-ddr3-330.dtbo
│   ├── rk322x-ddr3-528.dtbo
│   ├── rk322x-ddr3-660.dtbo
│   ├── rk322x-ddr3-800.dtbo
│   ├── rk322x-emmc-ddr-ph180.dtbo
│   ├── rk322x-emmc-ddr-ph45.dtbo
│   ├── rk322x-emmc.dtbo
│   ├── rk322x-emmc-hs200.dtbo
│   ├── rk322x-emmc-pins.dtbo
│   ├── rk322x-fixup.scr
│   ├── rk322x-ir-wakeup.dtbo
│   ├── rk322x-led-conf1.dtbo
│   ├── rk322x-led-conf2.dtbo
│   ├── rk322x-led-conf3.dtbo
│   ├── rk322x-led-conf4.dtbo
│   ├── rk322x-led-conf5.dtbo
│   ├── rk322x-led-conf6.dtbo
│   ├── rk322x-led-conf7.dtbo
│   ├── rk322x-led-conf8.dtbo
│   ├── rk322x-led-conf-default.dtbo
│   ├── rk322x-nand.dtbo
│   ├── rk322x-usb-otg-peripheral.dtbo
│   ├── rk322x-wlan-alt-wiring.dtbo
│   ├── rockchip-ds1307.dtbo
│   ├── rockchip-fixup.scr
│   ├── rockchip-i2c1.dtbo
│   ├── rockchip-i2c4.dtbo
│   ├── rockchip-pwm1.dtbo
│   ├── rockchip-pwm2.dtbo
│   ├── rockchip-pwm3.dtbo
│   ├── rockchip-spi0.dtbo
│   ├── rockchip-spi2.dtbo
│   ├── rockchip-spidev0.dtbo
│   ├── rockchip-spidev2.dtbo
│   ├── rockchip-uart1.dtbo
│   ├── rockchip-uart2.dtbo
│   ├── rockchip-uart3.dtbo
│   ├── rockchip-uart4.dtbo
│   └── rockchip-w1-gpio.dtbo
├── rk3036-evb.dtb
├── rk3036-kylin.dtb
├── rk3066a-bqcurie2.dtb
├── rk3066a-marsboard.dtb
├── rk3066a-mk808.dtb
├── rk3066a-rayeager.dtb
├── rk3128-evb.dtb
├── rk3188-bqedison2qc.dtb
├── rk3188-px3-evb.dtb
├── rk3188-radxarock.dtb
├── rk3228-evb.dtb
├── rk3229-evb.dtb
├── rk3229-xms6.dtb
├── rk322x-box.dtb
├── rk3288-evb-act8846.dtb
├── rk3288-evb-rk808.dtb
├── rk3288-firefly-beta.dtb
├── rk3288-firefly.dtb
├── rk3288-firefly-reload.dtb
├── rk3288-miqi.dtb
├── rk3288-phycore-rdk.dtb
├── rk3288-popmetal.dtb
├── rk3288-r89.dtb
├── rk3288-rock2-square.dtb
├── rk3288-rock-pi-n8.dtb
├── rk3288-tinker.dtb
├── rk3288-tinker-s.dtb
├── rk3288-veyron-brain.dtb
├── rk3288-veyron-fievel.dtb
├── rk3288-veyron-jaq.dtb
├── rk3288-veyron-jerry.dtb
├── rk3288-veyron-mickey.dtb
├── rk3288-veyron-mighty.dtb
├── rk3288-veyron-minnie.dtb
├── rk3288-veyron-pinky.dtb
├── rk3288-veyron-speedy.dtb
├── rk3288-veyron-tiger.dtb
├── rk3288-vyasa.dtb
├── rk3288-xt-q8l-v10.dtb
├── rv1108-elgin-r1.dtb
├── rv1108-evb.dtb
└── rv1126-edgeble-neu2-io.dtb

1 directory, 87 files
verbosity=1
extraargs=coherent_pool=2M
bootlogo=false
overlay_prefix=rockchip
overlays=es90x8q2m-dac,i2c1
rootdev=UUID=a39b8cef-c3a4-41d8-a0cf-fe08d139f982
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u

`

[image]

igorpecovnik, 2025-01-25:
> Corrected but also the overlays are not shown.

Probably because you manually added es90x8q2m-dac which does not exists?

The-going, 2025-01-25:
> overlay_prefix=rockchip
> overlays=es90x8q2m-dac,i2c1

Valid:
```
overlay_prefix=rockchip
overlays=i2c1
```
**_overlays=name1 name2 name3 name4_**
names are separated by a space. The name must exist.

sp2ong, 2025-01-25:
> > This looks like an error in the armbian config script.
> 
> Am I blind, but I don't see any problems here. Prefix is not touched - it has to set right at build time.
> 
> https://github.com/armbian/configng/blob/main/tools/modules/system/manage_dtoverlays.sh#L48-L50

Hi, old armbin-config not NG when we go to select hardware show as all modules where names was without overlay_prefix and show only modules which has overlay prefix. When we select  modules and SAVE in armbianEnv.txt overlyas= was with overlays prefix. The armbian-congig NG new when we select SYS210 "Manage device tree overlays" show all modules where names included overlay prefix so I suppose for this reason when we select modules and SAVE in armbiaEnv.txt in overlays= are saved modules with overlay prefix for example sun8i-h3-analog-code but shuld be with overlay prefix

[image]

sp2ong, 2025-01-25:
According code: https://github.com/armbian/configng/blob/main/tools/modules/system/manage_dtoverlays.sh#L48-L50

`newoverlays=$(echo $selection | sed 's/"//g')`

simple test if name is **sun8i-h3-analog-codec**  result this code:

echo sun8i-h3-analog-codec |sed 's/"//g'
**sun8i-h3-analog-codec**

so this name is stored in armbianEnv.txt

but I suppose should be **analog-codec**

sp2ong, 2025-01-25:
Look on old armbian-config:  https://github.com/armbian/config/blob/473f9990fe722bc7fa2c77795f50dea3118708ff/debian-config-jobs#L1210
It looks like during load overlays to display in menu was striped overlay_prefix (defined in armbianEnv.txt for this reason displayed overlays was without prefix so if this name was without prefix it was stored in armbianEnv.txt in overlays= without overlay prefix and all works

For me when i armbian-config NG should be prepared list of overlays similar and should be display only overlays which has overla_prefix in name overlay

but I am not programs

sp2ong, 2025-01-25:
When I use code with old armbian-config to prepare list of overlay

def_overlays=$(ls -1 ${OVERLAYDIR}/${overlay_prefix}*.dtbo | sed 's/^.*\('${overlay_prefix}'.*\)/\1/g' | sed 's/'${overlay_prefix}'-//g' | sed 's/.dtbo//g')

result for overlay_prefix=sun8i-h3

ls -1 /boot/dtb/overlay/sun8i-h3*.dtbo | sed 's/^.*\('sun8i-h3'.*\)/\1/g' | sed 's/'sun8i-h3'-//g' | sed 's/.dtbo//g'
analog-codec
cir
cpu-clock-1.2GHz-1.3v
cpu-clock-1.368GHz-1.3v
cpu-clock-1.3GHz-1.3v
i2c0
i2c1
i2c2
pps-gpio
pwm
spdif-out
spi-add-cs1
spi-jedec-nor
spi-spidev
tve
uart1
uart2
uart3
usbhost0
usbhost1
usbhost2
usbhost3
w1-gpio

so we have overlays names without overlay prefix and displayed only for sun8i-h3

sp2ong, 2025-01-25:
In new armbian-config list of overlays is created

`available_overlays=$(ls -1 ${overlaydir}/*.dtbo | sed "s#^${overlaydir}/##" | sed 's/.dtbo//g' | grep -E "$BOOT_SOC|$BOARD" | tr '\n' ' ')`

so result this code is

ls -1 /boot/dtb/overlay/*.dtbo | sed "s#^/boot/dtb/overlay/##" | sed 's/.dtbo//g' | grep -E "$BOOT_SOC|$BOARD" | tr '\n' ' '
sun4i-a10-analog-codec sun4i-a10-can sun4i-a10-i2c1 sun4i-a10-i2c2 sun4i-a10-nand sun4i-a10-pps-gpio sun4i-a10-pwm sun4i-a10-spdif-out sun4i-a10-spi-jedec-nor sun4i-a10-spi-spidev sun4i-a10-uart2 sun4i-a10-uart3 sun4i-a10-uart4 sun4i-a10-uart5 sun4i-a10-uart6 sun4i-a10-uart7 sun4i-a10-w1-gpio sun5i-a13-analog-codec sun5i-a13-i2c1 sun5i-a13-i2c2 sun5i-a13-nand sun5i-a13-pwm sun5i-a13-spi0 sun5i-a13-spi1 sun5i-a13-spi2 sun5i-a13-spi-jedec-nor sun5i-a13-spi-spidev sun5i-a13-uart0 sun5i-a13-uart1 sun5i-a13-uart2 sun5i-a13-uart3 sun7i-a20-analog-codec sun7i-a20-can sun7i-a20-i2c1 sun7i-a20-i2c2 sun7i-a20-i2c3 sun7i-a20-i2c4 sun7i-a20-mmc2 sun7i-a20-nand sun7i-a20-pps-gpio sun7i-a20-pwm sun7i-a20-spdif-out sun7i-a20-spi-add-cs1 sun7i-a20-spi-jedec-nor sun7i-a20-spi-spidev sun7i-a20-uart2 sun7i-a20-uart3 sun7i-a20-uart4 sun7i-a20-uart5 sun7i-a20-uart6 sun7i-a20-uart7 sun7i-a20-w1-gpio sun8i-h3-analog-codec sun8i-h3-cir sun8i-h3-cpu-clock-1.2GHz-1.3v sun8i-h3-cpu-clock-1.368GHz-1.3v sun8i-h3-cpu-clock-1.3GHz-1.3v sun8i-h3-i2c0 sun8i-h3-i2c1 sun8i-h3-i2c2 sun8i-h3-pps-gpio sun8i-h3-pwm sun8i-h3-spdif-out sun8i-h3-spi-add-cs1 sun8i-h3-spi-jedec-nor sun8i-h3-spi-spidev sun8i-h3-tve sun8i-h3-uart1 sun8i-h3-uart2 sun8i-h3-uart3 sun8i-h3-usbhost0 sun8i-h3-usbhost1 sun8i-h3-usbhost2 sun8i-h3-usbhost3 sun8i-h3-w1-gpio sun8i-r40-i2c2 sun8i-r40-i2c3 sun8i-r40-spi-spidev0 sun8i-r40-spi-spidev1 sun8i-r40-uart2 sun8i-r40-uart4 sun8i-r40-uart5 sun8i-r40-uart7

long list all overlays which we see in menu overlays in armbian-config, I am not sure about grep -E "$BOOT_SOC|$BOARD"   where is defined

igorpecovnik, 2025-01-25:
Thank you for help. I think this is it, but need more testing: https://github.com/armbian/configng/pull/393

sp2ong, 2025-01-25:
Ok no problem, it looks like new armbian-config not use overlay_prefix= from amrmbianEnv.txt which help to reduce displayed overlays to manage and strip this name in overlays. If you create new version armbin-config I will be check.

Regards
Waldek

sp2ong, 2025-01-25:
Ah I see variable $BOARD_SOC and $BOARD is coming from /etc/armbian-image-release but on Orange Pi Zero BOARD_SOC is empty and BOARD is orangepizro but it is not match in names in overlays so old method which used overlay_prefox form amrmbianEnv.txt works quite good according to doc ArmBian: https://docs.armbian.com/User-Guide_Armbian_overlays/  `overlay_prefix - prefix for the DT and overlay file names, set at OS image creation time`

_if [[ -n "${BOOT_SOC}" ]]; then
available_overlays=$(ls -1 ${overlaydir}/*.dtbo | sed "s#^${overlaydir}/##" | sed 's/.dtbo//g' | grep -E "$BOOT_SOC|$BOARD" | tr '\n' ' ')
else
available_overlays=$(ls -1 ${overlaydir}/*.dtbo | sed "s#^${overlaydir}/##" | sed 's/.dtbo//g' | tr '\n' ' ')
 fi_

igorpecovnik, 2025-01-26:
This should be fixed now, closing. Reopen in case of troubles.

closed 2025-01-26 by igorpecovnik

MMinga, 2025-01-28:
It does still not work on my system, where is my fault, if so?
HW: tinkerboard 
```/boot/dtb/
├── overlay
│   ├── README.rk322x-overlays
│   ├── README.rockchip-overlays
│   ├── rk322x-bt-8723cs.dtbo
│   ├── rk322x-cpu-hs.dtbo
│   ├── rk322x-cpu-hs-lv.dtbo
│   ├── rk322x-cpu-stability.dtbo
│   ├── rk322x-ddr3-330.dtbo
│   ├── rk322x-ddr3-528.dtbo
│   ├── rk322x-ddr3-660.dtbo
│   ├── rk322x-ddr3-800.dtbo
│   ├── rk322x-emmc-ddr-ph180.dtbo
│   ├── rk322x-emmc-ddr-ph45.dtbo
│   ├── rk322x-emmc.dtbo
│   ├── rk322x-emmc-hs200.dtbo
│   ├── rk322x-emmc-pins.dtbo
│   ├── rk322x-fixup.scr
│   ├── rk322x-ir-wakeup.dtbo
│   ├── rk322x-led-conf1.dtbo
│   ├── rk322x-led-conf2.dtbo
│   ├── rk322x-led-conf3.dtbo
│   ├── rk322x-led-conf4.dtbo
│   ├── rk322x-led-conf5.dtbo
│   ├── rk322x-led-conf6.dtbo
│   ├── rk322x-led-conf7.dtbo
│   ├── rk322x-led-conf8.dtbo
│   ├── rk322x-led-conf-default.dtbo
│   ├── rk322x-nand.dtbo
│   ├── rk322x-usb-otg-peripheral.dtbo
│   ├── rk322x-wlan-alt-wiring.dtbo
│   ├── rockchip-ds1307.dtbo
│   ├── rockchip-es90x8q2m-dac.dtbo
│   ├── rockchip-fixup.scr
│   ├── rockchip-i2c1.dtbo
│   ├── rockchip-i2c4.dtbo
│   ├── rockchip-pwm1.dtbo
│   ├── rockchip-pwm2.dtbo
│   ├── rockchip-pwm3.dtbo
│   ├── rockchip-spi0.dtbo
│   ├── rockchip-spi2.dtbo
│   ├── rockchip-spidev0.dtbo
│   ├── rockchip-spidev2.dtbo
│   ├── rockchip-uart1.dtbo
│   ├── rockchip-uart2.dtbo
│   ├── rockchip-uart3.dtbo
│   ├── rockchip-uart4.dtbo
│   └── rockchip-w1-gpio.dtbo
├── rk3036-evb.dtb
├── rk3036-kylin.dtb
├── rk3066a-bqcurie2.dtb
├── rk3066a-marsboard.dtb
├── rk3066a-mk808.dtb
├── rk3066a-rayeager.dtb
├── rk3128-evb.dtb
├── rk3188-bqedison2qc.dtb
├── rk3188-px3-evb.dtb
├── rk3188-radxarock.dtb
├── rk3228-evb.dtb
├── rk3229-evb.dtb
├── rk3229-xms6.dtb
├── rk322x-box.dtb
├── rk3288-evb-act8846.dtb
├── rk3288-evb-rk808.dtb
├── rk3288-firefly-beta.dtb
├── rk3288-firefly.dtb
├── rk3288-firefly-reload.dtb
├── rk3288-miqi.dtb
├── rk3288-phycore-rdk.dtb
├── rk3288-popmetal.dtb
├── rk3288-r89.dtb
├── rk3288-rock2-square.dtb
├── rk3288-rock-pi-n8.dtb
├── rk3288-tinker.dtb
├── rk3288-tinker-s.dtb
├── rk3288-veyron-brain.dtb
├── rk3288-veyron-fievel.dtb
├── rk3288-veyron-jaq.dtb
├── rk3288-veyron-jerry.dtb
├── rk3288-veyron-mickey.dtb
├── rk3288-veyron-mighty.dtb
├── rk3288-veyron-minnie.dtb
├── rk3288-veyron-pinky.dtb
├── rk3288-veyron-speedy.dtb
├── rk3288-veyron-tiger.dtb
├── rk3288-vyasa.dtb
├── rk3288-xt-q8l-v10.dtb
├── rv1108-elgin-r1.dtb
├── rv1108-evb.dtb
└── rv1126-edgeble-neu2-io.dtb

1 directory, 88 files
verbosity=1
extraargs=coherent_pool=2M
bootlogo=false
overlay_prefix=rockchip
overlays=es90x8q2m-dac i2c1
rootdev=UUID=a39b8cef-c3a4-41d8-a0cf-fe08d139f982
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u

apex-cmyk, 2025-02-18:
You can have a try of the Yajuzhen CloudPhone（亚矩阵云手机）. They are now using RK3588.