armbian/configng #360: [Bug]: Overlays aren't being added properly

MeestorX, 2025-01-06:
### What happened?

Using the latest armbian-config, if you select overlays to be added, it puts them into armbianEnv.txt, but it's not respecting the overlay_prefix= setting. It's adding the full name of the overlay, and then the boot loader is prefixing that name with the overlay_prefix.
Deleting the overlay_prefix= line in armbianEnv.txt does not fix the problem, because it still prefixes a "-" ahead of the name.

### How to reproduce?

Examples and more information here: https://forum.armbian.com/topic/47793-overlays-do-not-apply-even-when-selected-in-armbian-config-and-then-rebooting/#findComment-210335

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MeestorX, 2025-01-06:
I ran apt update and apt upgrade, do I need to do something else to get the latest?

igorpecovnik, 2025-01-06:
Then this is a bug, probably beyond this config program and we will look into in when possible. I think there were some mentioning on how to fix on forums, various hardware. There are some troubles with overlays, but don't know if all were fixed. I think on the armbian-config side it was this https://github.com/armbian/configng/pull/310

MeestorX, 2025-01-06:
And maybe this: [#310](https://github.com/armbian/configng/pull/299)

igorpecovnik, 2025-01-06:
Oh yes, that. But that should fix it, hmm.

MeestorX, 2025-01-06:
The fix currently (as I put in the forum post) is to edit the overlay names in armbianEnv.txt, removing the prefix.

igorpecovnik, 2025-01-06:
Yeah, this is build issue. Image is from here, right: https://www.armbian.com/radxa-rock-s0/

sp2ong, 2025-01-16:
I confirm a similar problem on Orange Pi Zero 3 with the updated system on

_Welcome to Armbian_community 25.2.0-trunk.331 Bookworm Kernel Linux 6.6.70-current-sunxi64_

sp2ong, 2025-01-18:
I checked Orange Pi Zero v1 used latest minimal image Bookworm with updated system

_v25.2 rolling for Orange Pi Zero running Armbian Linux 6.6.70-current-sunxi_
_Packages:     Debian stable (bookworm)_
_Support:      for advanced users (rolling release)_

and exist problem when we use armbian-config and select Manage overlays where add to name overlay prefix so  user must manual remove prefix from armbianEnv.txt

igorpecovnik, 2025-01-23:
https://github.com/armbian/configng/issues/344

igorpecovnik, 2025-01-26:
@sp2ong @MeestorX try now. (apt update + upgrade to get latest armbian-config)

closed 2025-01-26 by igorpecovnik

sp2ong, 2025-01-26:
OK on Orange Pi Zero V1 now it works very well  :-)

[image]

sp2ong, 2025-01-26:
OK on Orange Pi Zero 3 now it works very well, but only one thing is strange for me, why does it additionally display bananapi-m4 overlay on Orange Pi Zero 3? Banana pi M4 use overlay prefix  sun50i-h616

Orange Pi Zero 3 armbianEnv.txt
```

verbosity=1
bootlogo=false
console=both
disp_mode=1920x1080p60
overlay_prefix=sun50i-h616
user_overlays=w1-gpio
rootdev=UUID=e67d17d6-3ccc-4b64-817a-ecbb6f368e47
rootfstype=ext4
overlays=i2c3-ph uart5-ph
param_w1_pin=PC10
param_w1_pin_int_pullup=0
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
```

[image]

[image]

igorpecovnik, 2025-01-26:
Great!

> why does it additionally display bananapi-m4 overlay

Because it also filter on board. If overlays is specific for one hardware, it might not work on another.

sp2ong, 2025-01-26:
Ah Ok now is clear me for Orange Pi Zero 3.
Many thanks for fixing the problem of display overlays in armbian-config in current images

Regards

Waldek

rivolity, 2025-02-03:
@sp2ong what config worked for you ? I have the same chip `orange pi zero 3`
can you share please the `armbianEnv.txt` file

sp2ong, 2025-02-03:
> [@sp2ong](https://github.com/sp2ong) what config worked for you ? I have the same chip `orange pi zero 3` can you share please the `armbianEnv.txt` file

My armbianEnv.txt:

```

verbosity=1
bootlogo=false
console=both
disp_mode=1920x1080p60
overlay_prefix=sun50i-h616
rootdev=UUID=d8d47429-0065-4c1b-98c4-3ba21da2f786
rootfstype=ext4
overlays=i2c3-ph uart5-ph
```

rivolity, 2025-02-03:
It does not work for me, I have used the same config as you, hardware does not detect.
@sp2ong 

[image]

sp2ong, 2025-02-03:
I use image based on

https://www.armbian.com/orange-pi-zero-3/

**Minimal/IOT images with Armbian Linux v6.6**

where is running

apt update
apt upgrade

I on my system is kernel 6.6.72

sp2ong, 2025-02-03:
uname -a
Linux hotspot-fmpl 6.6.72-current-sunxi64 #1 SMP Fri Jan 17 12:36:27 UTC 2025 aarch64 GNU/Linux

[image]

rivolity, 2025-02-03:
Do you have the hardware section ? When you select system ?
I'm using the desktop version.
@sp2ong

sp2ong, 2025-02-03:
At new armbian-config not exist HARDWARE all was moved to Kernel > Manage device tree

[image]

[image]

[image]

rivolity, 2025-02-03:
Oh my bad I can see it now, they have moved the hardware to the kernel, Manage device tree overlay. I can see it now, thank you so much.

[image]