armbian/configng #361: [Bug]: motd is blank, regardless of setting

MeestorX, 2025-01-06:
### What happened?

Not sure if this is an actual armbian-config issue or something else, but my motd is always blank, no matter what I've set in armbian-config.

### How to reproduce?

I feel like this issue may have started with 24.11.x, but not sure. I no longer see a motd on boot, so I tried adjusting it with armbian-config, but nothing seems to fix it.

armbianmonitor -u link: https://paste.next.armbian.com/palubuyima

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-01-06:
What do you have here:

ls -l /etc/update-motd.d 
total 36
```
-rwxr-xr-x 1 root root 2763 Feb 29  2024 10-armbian-header
-rwxr-xr-x 1 root root 6698 Feb 29  2024 30-armbian-sysinfo
-rwxr-xr-x 1 root root 1019 Feb 29  2024 35-armbian-tips
-rwxr-xr-x 1 root root 1252 Feb 29  2024 40-armbian-updates
-rwxr-xr-x 1 root root  895 Feb 29  2024 41-armbian-config
-rwxr-xr-x 1 root root  165 Feb 19  2021 92-unattended-upgrades
-rwxr-xr-x 1 root root  554 Feb 29  2024 98-armbian-autoreboot-warn

```
Try
sudo chmod +x /etc/updated-motd.d/*

igorpecovnik, 2025-01-06:
You put it on forum, no need to put another ticket in build FW.

We will pay attention to it. I checked briefly in a code, but none of the areas that could have cause this hasn't been changed for a while. Must be some strange anomaly. It is not armbian-config, closing.

closed 2025-01-06 by igorpecovnik