armbian/configng #368: [Bug]: Disabling firmware upgrades broken?

EvilOlaf, 2025-01-11:
### What happened?

https://forum.armbian.com/topic/49253-correct-procedure-to-disable-kernel-updates/#comment-210648

### How to reproduce?

apparently only armbian-firmware package is put on hold but not dtb, kernel and headers

### On which host OS are you running the build script and observing this problem?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-01-11:
Could not reproduce with sunxi64. Did not test with rk35xx, maybe functionality was lost when merging this into rockchip64? Pure guess...

igorpecovnik, 2025-01-11:
Strange. We need to look into, yes.

vilaxara, 2025-01-18:
I am facing a similar issue. On rock 5 irx board with bookworm os image, I am using a 6.1.75-vendor-rk35xx kernal. After changing the option using armbian-config, when I do apt update and apt upgrade, it is updating it to 6.1.8*(I don'trememberthis nimber)-vendor-rk35xx. Is this supposed to happen?

igorpecovnik, 2025-01-22:
Steps to reproduce:

Freezing:
`armbian-config --cmd SY203`

Checking:
```
$ apt-mark showhold
armbian-firmware
linux-dtb-current-rockchip64
linux-image-current-rockchip64
```

Unfreezing:
`armbian-config --cmd SY202`

Checking:

```
$ apt-mark showhold
$
```

Works as expected. Make sure you have latest armbian-config package.

vilaxara, 2025-01-24:
Board type : Rock 5 ITX
OS : Debian 12 bookworm 24.11
Kernal : Custom kernal built with LSI modules enabled

`Linux rock-5-itx 6.1.75-vendor-rk35xx #1 SMP Wed Nov 27 22:16:11 EST 2024 aarch64 GNU/Linux`

armbian-config version : 

`$ dpkg -l | grep armbian-config`
`ii  armbian-config                          24.11.1                             all          Armbian configuration utility - The new generation`

I updated the armbian-config using :

`$ apt update && apt install --reinstall armbian-config`

`$ dpkg -l | grep armbian-config`
`ii  armbian-config                          25.2.0-trunk.378.0123.223643        all          Armbian config: The Next Generation
`
Rebooted the system

Everything is working fine where my raid card is getting dectected with `lspci -v`:

`0000:01:00.0 RAID bus controller: Broadcom / LSI MegaRAID SAS-3 3108 [Invader] (rev 02)`
`        Subsystem: Broadcom / LSI MegaRAID SAS 9364-8i`
`        Flags: bus master, fast devsel, latency 0, IRQ 133`
`        I/O ports at 200000 [disabled] [size=256]`
`        Memory at f0400000 (64-bit, non-prefetchable) [size=64K]`
`        Memory at f0200000 (64-bit, non-prefetchable) [size=1M]`
`        Expansion ROM at f0300000 [virtual] [disabled] [size=1M]`
`        Capabilities: [50] Power Management version 3`
`        Capabilities: [68] Express Endpoint, MSI 00`
`        Capabilities: [d0] Vital Product Data`
`        Capabilities: [a8] MSI: Enable- Count=1/1 Maskable+ 64bit+`
`        Capabilities: [c0] MSI-X: Enable+ Count=97 Masked-`
`        Capabilities: [100] Advanced Error Reporting`
`        Capabilities: [1e0] Secondary PCI Express`
`        Capabilities: [1c0] Power Budgeting <?>`
`        Capabilities: [148] Alternative Routing-ID Interpretation (ARI)`
`        Kernel driver in use: megaraid_sas`
`        Kernel modules: megaraid_sas`

Now I am following the given steps :

`$ armbian-config --cmd SY203`

Then :

`$ apt-mark showhold`
`armbian-firmware`

If I proceed to upgrade the kernal is getting updated to : `6.1.84-vendor-rk35xx` and `lspci -v` gives :

`0000:01:00.0 RAID bus controller: Broadcom / LSI MegaRAID SAS-3 3108 [Invader] (rev 02)`
`        Subsystem: Broadcom / LSI MegaRAID SAS 9364-8i`
`        Flags: bus master, fast devsel, latency 0, IRQ 133`
`        I/O ports at 200000 [disabled] [size=256]`
`        Memory at f0400000 (64-bit, non-prefetchable) [size=64K]`
`        Memory at f0200000 (64-bit, non-prefetchable) [size=1M]`
`        Expansion ROM at f0300000 [virtual] [disabled] [size=1M]`
`        Capabilities: [50] Power Management version 3`
`        Capabilities: [68] Express Endpoint, MSI 00`
`        Capabilities: [d0] Vital Product Data`
`        Capabilities: [a8] MSI: Enable- Count=1/1 Maskable+ 64bit+`
`        Capabilities: [c0] MSI-X: Enable+ Count=97 Masked-`
`        Capabilities: [100] Advanced Error Reporting`
`        Capabilities: [1e0] Secondary PCI Express`
`        Capabilities: [1c0] Power Budgeting <?>`
`        Capabilities: [148] Alternative Routing-ID Interpretation (ARI)`

The kernel drivers are not getting recognized.
This is happening for a long time.

igorpecovnik, 2025-01-24:
> $ apt-mark showhold
> armbian-firmware

This is not frozen kernel. How did you do this part?

vilaxara, 2025-01-24:
I used the following command :

armbian-config --cmd SY203

Even when I manually go to the interface and disable it (using sudo
armbian-config), it doesn't work.

domfe, 2025-01-25:
Looking at the code, seems BRANCH variable is required.

I added `BRANCH=vendor` in `/etc/armbian-config` and armbian-config work as expected

igorpecovnik, 2025-01-25:
BRANCH variable is set at boot time. Did you perhaps changing kernels from current to vendor and / or vice versa before that?

domfe, 2025-01-25:
I have no BRANCH set.
No. I stayed with kernel provided in the armbian image

vilaxara, 2025-01-25:
I can't find any `armbian-config` file.
```
$ ls /etc/armbian*
/etc/armbian-distribution-status  /etc/armbian-image-release  /etc/armbian-leds.conf  /etc/armbian-release  /etc/armbian.txt

/etc/armbian:
chromium  chromium.conf  firefox.conf  lightdm

/etc/armbianmonitor:
datasources
````

igorpecovnik, 2025-01-25:
> /etc/armbian-release

here you need to add RELEASE=

vilaxara, 2025-01-25:
Should I add RELEASE=vendor?

I edited the file as such : 

```
# PLEASE DO NOT EDIT THIS FILE
BOARD=rock-5-itx
BOARD_NAME="Rock 5 ITX"
BOARDFAMILY=rockchip-rk3588
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=8785ea758
LINUXFAMILY=rk35xx
ARCH=arm64
BOOT_SOC=rk3588
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
KERNEL_TARGET=vendor
KERNEL_TEST_TARGET=
FORCE_BOOTSCRIPT_UPDATE=
FORCE_UBOOT_UPDATE=
OVERLAY_DIR="/boot/dtb/rockchip/overlay"
VENDOR="Armbian-unofficial"
VENDORDOCS="https://docs.armbian.com/"
VENDORURL="https://duckduckgo.com/"
VENDORSUPPORT="https://community.armbian.com/"
VENDORBUGS="https://armbian.atlassian.net/"
BOOTSCRIPT_FORCE_UPDATE="no"
BOOTSCRIPT_DST="boot.cmd"
VERSION=24.11.1
REVISION=24.11.1
RELEASE=vendor
```

It did not work

vilaxara, 2025-01-25:
By adding the lines `RELEASE=bookworm`, `BRANCH=vendor` to armbian-release file and proceeded with the steps after rebooting the system. It worked :

```
$ armbian-config --cmd SY203 

$ apt-mark showhold
 armbian-firmware
linux-dtb-vendor-rk35xx
linux-image-vendor-rk35xx
```

I will upgrade and post the update

vilaxara, 2025-01-25:
It works. 🎉

igorpecovnik, 2025-01-25:
OK, great. I will implement some automation here to prevent this corners case to happen. Lets keep this open.

closed 2025-02-07 by igorpecovnik