armbian/configng #376: [Bug]: armbian-config stuck without CLI configuration GUI

edwardzcn-decade, 2025-01-18:
### What happened?

I try to reinstall the `armbian-config` but it seems no help. I cant see the panel in CLI.

[image]

Here is my terminal:

```bash
echo $TERM
xterm-256color
```

and os-release

```bash
edwardzcn@armsom-sige5:/etc/apt/sources.list.d$ cat /etc/os-release
PRETTY_NAME="Armbian 24.11.3 noble"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.armbian.com"
SUPPORT_URL="https://forum.armbian.com"
BUG_REPORT_URL="https://www.armbian.com/bugs"
PRIVACY_POLICY_URL="https://www.armbian.com"
UBUNTU_CODENAME=noble
LOGO="armbian-logo"
ARMBIAN_PRETTY_NAME="Armbian 24.11.3 noble"
```

### How to reproduce?

I use Armbian on a armsom-sige5 board. The motd shows that the armbian OS is based on Ubuntu 24.04 noble

```txt
Welcome to Armbian-unofficial 24.8.0-trunk Noble with Linux 6.1.57-rk3576-rk35xx
```

Just apt update and apt upgrade and the armbian-config not work.

### On which host OS are you running the build script and observing this problem?

Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-01-22:
> unofficial 24.8.0-trunk Noble with Linux 6.1.57-rk3576-rk35xx

It is not possible to reproduce / possible hardware related, which is outside this ticketing system @ac-jackson

edwardzcn-decade, 2025-01-24:
I guess it's because of internet problems. Still thx for your kind help.

closed 2025-01-24 by edwardzcn-decade