armbian/configng #385: [Bug]: armbian-config /localization/change Keyboard layout issue.

curyjorge98, 2025-01-21:
### What happened?

Running "armbian-config"/localization/change Keyboard layout, from a terminal window this msg is displayed:

"setupcon: We are not on the console, the console is left unconfigured."

But if I run "sudo dpkg-reconfigure keyboard-configuration" from a terminal window no error msg is displayed.

Installed Deb 12 Cinnamon Armbian flavor.
This: dl.armbian.com/orangepi5-plus/Bookworm_current_cinnamon-backported-mesa 

Computer: Orange PI 5 and Orange PI 5 PLUS

me@opi5-normal-cinn:~/Vídeos$ cat /etc/os-release
PRETTY_NAME="Armbian 24.11.3 bookworm"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.armbian.com"
SUPPORT_URL="https://forum.armbian.com"
BUG_REPORT_URL="https://www.armbian.com/bugs"
ARMBIAN_PRETTY_NAME="Armbian 24.11.3 bookworm"

Thanks,

Cury

### How to reproduce?

Run "armbian-config"/localization/change Keyboard layout, from a terminal window

### On which host OS are you running the build script and observing this problem?

Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-01-22:
> But if I run "sudo dpkg-reconfigure keyboard-configuration" from a terminal window no error msg is displayed.

This is armbian-config way:

https://github.com/armbian/configng/blob/main/tools/json/config.localisation.json#L31

And if you enter those two commands into the terminal, you will get this error too.

igorpecovnik, 2025-01-23:
https://github.com/armbian/configng/issues/288

curyjorge98, 2025-01-23:
Hi Igor,

But if I run "sudo dpkg-reconfigure keyboard-configuration" from a terminal window 
**no error msg is displayed.**

Thanks

igorpecovnik, 2025-01-23:
And if you run: `sudo setupcon` afterward?

curyjorge98, 2025-01-25:
When I run this commando on a gnome terminal:

**

> me@orangepi5-plus:~$ sudo setupcon
> setupcon: We are not on the console, the console is left unconfigured.

**

igorpecovnik, 2025-01-25:
What happens if you reboot after this? Does it change as expected?

curyjorge98, 2025-01-25:
Hi, after reboot things go on the same way.

But I found a way to run this "dpkg-reconfigure" commands without error. Typing "ctrl-alt-f1" no error is displayed.

But new warning are displayed near the end of "sudo dpkg-reconfigure keyboard..."

Are they normal???

update-initramfs: Generating /boot/initrd. img-6.1.84-vendor-rk35xx
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8125a-3.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl81e7e-2.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168fp-3.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rt18168g-3.fw for module r8169 
W: Possible missing firmware /1ib/firmware/rtl_nic/rtl8168g-2.fw for module r8169 
W: Possible missing firmware /lib/firmware/rti-nic/rtl8166e-2.fw for module r8169 
W: Possible missing firmware /1ib/firmware/rti-nic/rtlsle6e-1.fw for module r816o 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8411-2.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8411-1.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8492-1.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168f-2.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168f-1.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl81@5e-1.fw for module r8169 
W: Possible missing firmware/lib/firmware/rtl_nic/rtl8168e-3.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168e-2.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168e-1.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168d-2.fw for module r8169 
W: Possible missing firmware /lib/firmware/rtl_nic/rtl8168d-1.fw for module r8169

[image]

Thanks

igorpecovnik, 2025-01-25:
> Hi, after reboot things go on the same way.

OK, then I will add mandatory reboot here.

> Are they normal???

Yes, safe to ignore. But I was already thinking to add this to stop this wall of warnings.

closed 2025-01-26 by curyjorge98