armbian/configng #412: [Feature Request]: Add Zerotier software

jnovos, 2025-02-04:
### Which feature would you like to have?

Allows joining virtual networks as VPN connections on your Android phone, tablet or other orangepi :-)

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

igorpecovnik, 2025-02-04:
And probably also Tailscale which is something similar to this?

jnovos, 2025-02-19:
@igorpecovnik I have added to my armbian-config the code to install this software. To add it to the “config” repository I follow this tutorial? https://docs.armbian.com/Contribute/Armbian-config/#adding-software-title-to-armbian-config. Thanks

igorpecovnik, 2025-02-19:
> To add it to the “config” repository I follow this tutorial?

Yes!! Try it. It is also a test if those instructions are done well.