armbian/configng #414: Session crash on early exit from network set static

stephengraf, 2025-02-05:
### What happened?

sysadmin@orangepione:~$ sudo armbian-config
[sudo] password for sysadmin:
The current Armbian 25.02.0-trunk user-built (bookworm) is supported.
Loaded Runtime variables...
Loaded Dialog...
Loaded Docs...
Loaded System helpers...
Loaded Network helpers...
Loaded Software helpers...
Loaded Runtime conditions...

Used Network - Basic - select adapter - Configure - Spoof - Set ip . I think I entered an invalid route ip and after entering the a gateway ip I cancelled out.  Following is the  session crash on cancelling. The system was still operational via the serial port but this ssh login session was unresponsive.

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.745: g_file_new_for_path: assertion 'path != NU   LL' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.745: g_file_read: assertion 'G_IS_FILE (file)'    failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.745: g_data_input_stream_new: assertion 'G_IS_I   NPUT_STREAM (base_stream)' failed

(process:2032): GLib-GObject-CRITICAL **: 11:59:10.746: g_object_unref: assertion 'G_IS_OBJECT    (object)' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.746: g_data_input_stream_read_line: assertion '   G_IS_DATA_INPUT_STREAM (stream)' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.746: g_data_input_stream_read_line: assertion '   G_IS_DATA_INPUT_STREAM (stream)' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.746: g_data_input_stream_read_line: assertion '   G_IS_DATA_INPUT_STREAM (stream)' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.746: g_data_input_stream_read_line: assertion '   G_IS_DATA_INPUT_STREAM (stream)' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.746: g_data_input_stream_read_line: assertion '   G_IS_DATA_INPUT_STREAM (stream)' failed

(process:2032): GLib-GIO-CRITICAL **: 11:59:10.746: g_data_input_stream_read_line: assertion '   G_IS_DATA_INPUT_STREAM (stream)' failed

(process:2032): GLib-GObject-CRITICAL **: 11:59:10.746: g_object_unref: assertion 'G_IS_OBJECT    (object)' failed

(process:2032): GLib-GObject-CRITICAL **: 11:59:10.746: g_object_unref: assertion 'G_IS_OBJECT    (object)' failed
Traceback (most recent call last):
  File "/usr/sbin/netplan", line 23, in <module>
    netplan.main()
  File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 50, in run
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 85, in command_set
    parser.load_yaml(tmp)
  File "/usr/share/netplan/netplan/libnetplan.py", line 119, in load_yaml
    _checked_lib_call(lib.netplan_parser_load_yaml_from_fd, self._ptr, input_file.fileno())
  File "/usr/share/netplan/netplan/libnetplan.py", line 79, in _checked_lib_call
    raise LibNetplanException(err.contents.message.decode('utf-8'))
netplan.libnetplan.LibNetplanException: (null):6:16: Error in network definition: invalid IP f   amily '-1'
(null)
               ^

### How to reproduce?

armbian-config
Network
Basic Network Setup
Select interface
Configure
Static Set IP Manually
Spoof (left default)
Enter IP (192.168.1.12/24)
Use default route (left default)
Enter IP for gateway (entered 192.168.1.2, which I think exists on another device)
Enter DNS server (used cancel at this point)

This time it did not crash but generated the following messages.
** (process:5586): WARNING **: 14:05:43.678: Problem encountered while validating default route consistency.Please set up multiple routing tables and use `routing-policy` instead.
Error: Conflicting default route declarations for IPv4 (table: main, metric: 200), first declared in end0 but also in end0

** (process:5586): WARNING **: 14:05:43.681: Problem encountered while validating default route consistency.Please set up multiple routing tables and use `routing-policy` instead.
Error: Conflicting default route declarations for IPv4 (table: main, metric: 200), first declared in end0 but also in end0

Now armbian-config no longer starts, even after a reboot and from the root account.

Problem with armbianmonitor also

root@orangepione:~# armbian-config
^Croot@orangepione:~# armbianmonitor -u
Collecting info and sending to paste.armbian.com, wait...
Failed grabbing info (pipe 3 result 28) and sending to server paste.armbian.com.
Collecting info and sending to paste.next.armbian.com, wait...

I might have to generate a new sd image???

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

stephengraf, 2025-02-05:
After rewriting the sd image, still got an error and then iarmbianmonitor -u worked.

root@orangepione:~# armbianmonitor -u
Collecting info and sending to paste.armbian.com, wait...
Failed grabbing info (pipe 3 result 22) and sending to server paste.armbian.com.
Collecting info and sending to paste.next.armbian.com, wait...

https://paste.next.armbian.com/wufepofupo

Please post the URL in the forum where you've been asked for.

root@orangepione:~#

igorpecovnik, 2025-02-05:
Thank you for testing.

> I think I entered an invalid route 

Yeah, we don't have a check for that yet. In case of wrong entry, netplan will crash. I need to check how this can be done right.

> still got an error and then iarmbianmonitor -u worked.

That is not yet fixed, but we are working on:
https://armbian.atlassian.net/browse/DEVOPS-136

stephengraf, 2025-02-05:
Thank you for your reply. The network config works if you put in proper values.

Many years ago I was given the task of writing an operators console for a process control system. When I finished it and tested I called my boss to show off my wonderful work. He came over, just leaned his hand on the keyboard keys and the system crashed! I had written a number of projects to do the process control of huge thundering paper making machines, but writing for the idiot at the keyboard was just a different world.