armbian/configng #421: [Bug]: setting up / reconfiguing interfaces

N4CNR, 2025-02-06:
### What happened?

when you go to networking >basic configuration > select the interface like wlan0 > drop its not dropping the interface config. and on a reboot it comes right back up with dhcp. If you try to resetup the interface it errors out . error goes by to fast to get a copy of it... but now stuckona system where its not allowing me to change settings onthe interface

### How to reproduce?

follow steps above

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pepihub, 2025-03-26:
Hi @igorpecovnik 

Since the op didn't went further, let me give it a try on this same thread, i'm having almost the same issue.

When trying to drop or reconfigure the wlan0 interface, from static to dhcp, doesn't matter, these are the logs i'm getting:

(the logs might be duplicated since i'm not sure where it starts)

```shell
Cannot call openvswitch: ovsdb-server.service is not running.

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_file_new_for_path: assertion 'path != NULL' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_file_read: assertion 'G_IS_FILE (file)' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_data_input_stream_new: assertion 'G_IS_INPUT_STREAM (base_stream)' failed

(process:3703): GLib-GObject-CRITICAL **: 18:05:47.796: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3703): GLib-GIO-CRITICAL **: 18:05:47.796: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3703): GLib-GObject-CRITICAL **: 18:05:47.796: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:3703): GLib-GObject-CRITICAL **: 18:05:47.796: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Traceback (most recent call last):
  File "/usr/sbin/netplan", line 23, in <module>
    netplan.main()
  File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 50, in run
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 85, in command_set
    parser.load_yaml(tmp)
  File "/usr/share/netplan/netplan/libnetplan.py", line 119, in load_yaml
    _checked_lib_call(lib.netplan_parser_load_yaml_from_fd, self._ptr, input_file.fileno())
  File "/usr/share/netplan/netplan/libnetplan.py", line 79, in _checked_lib_call
    raise LibNetplanException(err.contents.message.decode('utf-8'))
netplan.libnetplan.LibNetplanException: (null):5:9: Error in network definition: wlan0: Duplicate access point SSID 'CNI'
(null)
        ^

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.521: g_file_new_for_path: assertion 'path != NULL' failed

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.521: g_file_read: assertion 'G_IS_FILE (file)' failed

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.521: g_data_input_stream_new: assertion 'G_IS_INPUT_STREAM (base_stream)' failed

(process:3705): GLib-GObject-CRITICAL **: 18:05:48.521: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.522: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.522: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.522: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3705): GLib-GIO-CRITICAL **: 18:05:48.522: g_data_input_stream_read_line: assertion 'G_IS_DATA_INPUT_STREAM (stream)' failed

(process:3705): GLib-GObject-CRITICAL **: 18:05:48.522: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
[... 201 lines cut ...]
(process:3716): GLib-GObject-CRITICAL **: 18:05:52.131: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Traceback (most recent call last):
  File "/usr/sbin/netplan", line 23, in <module>
    netplan.main()
  File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 50, in run
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 241, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 120, in command_set
    parser_output_file.load_yaml(tmp)
  File "/usr/share/netplan/netplan/libnetplan.py", line 119, in load_yaml
    _checked_lib_call(lib.netplan_parser_load_yaml_from_fd, self._ptr, input_file.fileno())
  File "/usr/share/netplan/netplan/libnetplan.py", line 79, in _checked_lib_call
    raise LibNetplanException(err.contents.message.decode('utf-8'))
netplan.libnetplan.LibNetplanException: (null):4:7: Error in network definition: wlan0: No access points defined
(null)
      ^
Cannot call openvswitch: ovsdb-server.service is not running.
``` 

When modifying the netplan yaml file itself it works, but not via armbian-config.

igorpecovnik, 2025-03-26:
> When modifying the netplan yaml file itself it works, but not via armbian-config.

Understand. Can you provide:
- armbianmonitor -u
- steps to reproduce (from clean image)

pepihub, 2025-03-26:
https://paste.armbian.com/cidigohufi

> steps to reproduce (from clean image)

Armbian image built custom for orangepizero2w, extension preset-firstrun with wifi SSID and password.

Then I just login via SSH and run armbian-config, basic network, try to change the wlan to static instead of dhcp and the log of my previous message shows, then the armbian-config utility shows back with no changes whatsoever.

igorpecovnik, 2025-04-03:
This should be fixed now, closing.

pepihub, 2025-04-05:
> This should be fixed now, closing.

I've tried with the latest version and I still get the same errors as before.

Strat00s, 2025-05-17:
I am also getting the exact same errors on Orange Pi Zero 3.

BstAA, 2025-08-10:
Problem is still present :-(

See [armbian-monitor output ](https://paste.armbian.com/voxujemoca) and attached console output.

[Error-1.txt](https://github.com/user-attachments/files/21704661/Error-1.txt)

MaciejMalczyk, 2026-01-12:
Can be omitted by installing network-manager, enabling NetworkManager.service and setting connection with nmtui.