armbian/configng #431: [Bug]:

curyjorge98, 2025-02-10:
### What happened?

Hi everybody,

 

I installed Deb 12 Cinnamon Armbian flavor on an Orange PI 5 and Orange PI 5 Plus.

 This: dl.armbian.com/orangepi5-plus/Bookworm_current_cinnamon-backported-mesa 

When I open system setup there is no Idiom/Language Icon.  So all apps remains in English.

Even if I run "sudo dpkg-reconfigure keyboard-configuration" and "sudo dpkg-reconfigure locales"

I need pt-BR, 

This important config is missing on Armbian above.

armbianmonitor -u

[https://paste.next.armbian.com/egexohofom](url)

Thanks,

Cury 

### How to reproduce?

Just install this: dl.armbian.com/orangepi5-plus/Bookworm_current_cinnamon-backported-mesa 

on an Orange pi 5 or pi 5 Plus and nobody can change system language to Brazilian-Portuguese or Italian or French or whatever.

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-02-11:
This is beyond armbian-config tool. There must be a configuration in the Cinnamon desktop itself (could be missing), to setup that. 

[image]

This is my Armbian Gnome desktop, which is configured to English keyboard regardless of 1st boot locales. I need to add Slovene keyboard, then it works.

I am happy to integrate this to Cinnamon [desktop](https://github.com/armbian/build/tree/main/config/desktop), if you can find out what is missing.

curyjorge98, 2025-02-11:
Hi Igor,

I didn't know this place was only for armbian-config, sorry.

I have opened  this issue on Armbian Foruns almost 1 month ago, But I got no answer.

I am trying to find out the name of package fot seting up Language in cinnamon but no answer, including from Mint foruns.

I'll keep trying,

Thanks,

igorpecovnik, 2025-02-11:
No worries. I wish I had answer, but I would also need to go your path - asking around, searching, ... This is what AI tells: 

"The Cinnamon desktop environment was originally developed and is maintained by the Linux Mint team. The primary developer behind Cinnamon is Clément Lefèbvre, the founder and lead developer of Linux Mint. It was created as a fork of GNOME Shell to provide a more traditional desktop experience while maintaining modern features."

I would put my bet that its similar / same as in Gnome 2.

Tearran, 2025-05-01:
>I am trying to find out the name of package fot seting up Language in cinnamon but no answer, including from Mint foruns.

AI suggested 

`gnome-control-center region`

alternatively

 `language-selector-gnome` 

Cinnamon often relies on tools provided by the underlying operating system or desktop environment. These tools are usually part of GNOME or other system utilities.

igorpecovnik, 2025-07-15:
Closing as this is not a part of armbian-config but DE. Armbian sets locales at 1st boot - which precondition - and many desktops automatically adjust language based on that. Changing this is beyond Armbian too. We don't deal with desktop environments core functionality. This is done by upstream projets  / organizations such as https://en.wikipedia.org/wiki/GNOME , https://en.wikipedia.org/wiki/KDE and similars. Closing.