armbian/configng #449: [Bug]: armbian-config hangs

MoltenHydrogen, 2025-02-19:
### What happened?

On a fresh install. When I run armbian-config, the process hangs indefinitely. Running on an rk322x board.

https://paste.next.armbian.com/paxesoqomu Here are logs from `armbianmonitor -u`

### How to reproduce?

1. Be on rk322x
2. run `armbian-config`
3. wait...

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

MoltenHydrogen, 2025-02-19:
Nevermind, looks like it just took a really long time to load (like 20 minutes ish). Maybe add something letting know it'll take a while.

closed 2025-02-19 by MoltenHydrogen

igorpecovnik, 2025-02-19:
> looks like it just took a really long time to load (like 20 minutes ish)

That is not normal. Make sure to have latest version (apt update & apt upgrade) then see where it takes long time. This is on my desktop:

[image]

MoltenHydrogen, 2025-02-20:
Took 543 seconds to load runtime variables... Everything else loaded in 0-1 seconds.

igorpecovnik, 2025-02-20:
> Took 543 seconds to load runtime variables.

Wow. High IO on slow system would explain that. I never saw more then few seconds here doing many many runs.

> Be on rk322x

Also this target is totally unknown to me. Support for TV boxes is random and perhaps we have some strange combination of hardware that keep system resources in bad state ...