armbian/configng #458: [Bug]: "SY203 - Disable Armbian kernel upgrades" doesn't seem to take

henla464, 2025-02-28:
### What happened?

I have run "SY203 - Disable Armbian kernel upgrades"... and then after the requested reboot when I go into the System Update menu again the option is still there. It appears it doesn't disable the kernel upgrades, or if it does it doesn't recognize it has.

This is on a NanoPi Neo Air board.
v25.5 rolling for NanoPi Air running Armbian Linux 6.6.75-current-sunxi

### How to reproduce?

1) sudo armbian-config
2) go to System Update menu and select to "Disable Armbian kernel upgrades", confirm and reboot
3) after reboot go to System Update menu again, notice the option to "Disable Armbian kernel upgrades" is till there.

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

henla464, 2025-02-28:
armbianmonitor -u logs here: https://paste.next.armbian.com/adiriqafeq

Tearran, 2025-04-01:
including a name of the item to be held.

```bash
sudo apt-mark hold linux-image-generic
```

shows
```
linux-image-generic set on hold.
```

henla464, 2025-04-01:
I have another nanopi neo where 
`sudo apt-mark showhold`

shows
`armbian-firmware
linux-dtb-current-sunxi
linux-image-current-sunxi`

On this armbian-config shows "SY202 - Enable Armbian firmware upgrades" so that seems to work. I believe this is a slightly earlier image.

See https://paste.armbian.com/yoguyegehu

Tearran, 2025-04-02:
Thanks for report and updates. 

Is the solution for this issue 'update armbian-config'? 

'sudo apt update && sudo install armbian-config' 

To install the lasted bug fixes?

henla464, 2025-04-02:
Yes updating seem to fix it now!

closed 2025-04-02 by henla464