armbian/configng #480: [Bug]: armbian-config won't install docker engine

michelcastilho, 2025-03-17:
### What happened?

[armbianmonitor](https://paste.armbian.com/zuzenutese )

Docker-engine installation through armbian-config fails. Here's the log:

`Get:1 http://security.debian.org bookworm-security InRelease [48.0 kB]
Get:2 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 http://deb.debian.org/debian bookworm-backports InRelease [59.4 kB]
Ign:5 https://github.armbian.com/configng stable InRelease
Get:6 https://download.docker.com/linux/debian bookworm InRelease [43.3 kB]
Get:7 http://fi.mirror.armbian.de/beta bookworm InRelease [53.3 kB]
Ign:5 https://github.armbian.com/configng stable InRelease
Ign:5 https://github.armbian.com/configng stable InRelease
Err:5 https://github.armbian.com/configng stable InRelease
  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP: 2606:50c0:8003::153 443]
Reading package lists... Done
E: Release file for http://security.debian.org/dists/bookworm-security/InRelease is not valid yet (invalid for another 25d 9h 5min 11s). Updates for this repository will not be applied.
E: Release file for http://deb.debian.org/debian/dists/bookworm/InRelease is not valid yet (invalid for another 22d 22h 34min 0s). Updates for this repository will not be applied.
E: Release file for http://deb.debian.org/debian/dists/bookworm-updates/InRelease is not valid yet (invalid for another 25d 9h 41min 48s). Updates for this repository will not be applied.
E: Release file for http://deb.debian.org/debian/dists/bookworm-backports/InRelease is not valid yet (invalid for another 25d 9h 41min 48s). Updates for this repository will not be applied.
E: Release file for https://download.docker.com/linux/debian/dists/bookworm/InRelease is not valid yet (invalid for another 11d 0h 1min 59s). Updates for this repository will not be applied.
E: Release file for http://beta.armbian.com/dists/bookworm/InRelease is not valid yet (invalid for another 25d 7h 16min 17s). Updates for this repository will not be applied.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package docker-ce is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package docker-ce-cli is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'docker-ce' has no installation candidate
E: Package 'docker-ce-cli' has no installation candidate
E: Unable to locate package containerd.io
E: Couldn't find any package by glob 'containerd.io'
E: Couldn't find any package by regex 'containerd.io'
E: Unable to locate package docker-buildx-plugin
E: Unable to locate package docker-compose-plugin
Failed to enable unit: Unit file docker.service does not exist.
Failed to start docker.service: Unit docker.service not found.`

### How to reproduce?

Board:
OrangePi 3 LTS

OS running from eMMC:
Armbian_community_25.5.0-trunk.87_Orangepi3-lts_bookworm_current_6.6.72_minimal

apt updated and upgraded

Try to install Docker engine through armbian-config

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-03-17:
Probably your system date is invalid.

michelcastilho, 2025-03-19:
> Probably your system date is invalid.

Yep, that was the issue. Closing.

closed 2025-03-19 by michelcastilho