armbian/configng #482: [Bug]: armbian-* should have a man page and --help switch

leggewie, 2025-03-18:
### What happened?

Debian and Ubuntu users are used to having a man page for every program on the system.  I think armbian-* programs should have that, too.  I am happy to help in getting that done.

### How to reproduce?

man armbian-config

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

igorpecovnik, 2025-03-18:
armbian-config --help works:

```
Initializing script                     0 seconds
Loaded Runtime variables...             1 seconds
The current Armbian 25.2.2 stable (noble) is supported.           0 seconds
Loaded Dialog...                        0 seconds
Loaded Docs...                          0 seconds
Loaded System helpers...                0 seconds
Loaded Network helpers...               0 seconds
Loaded Software helpers...              0 seconds
Loaded Runtime conditions...            0 seconds
Usage:  armbian-config --[option]
  Options:
     --help [category]   Use [category] to filter specific menu options.
     --cmd  [option]     Run a command from the menu (simple)
     --api  [option]     Run a helper command        (advanced)
     --doc               Generate the README.md file 

     Examples:
     armbian-config --help [cmd||System||Software||Network||Localisation]
     armbian-config --cmd help 
     armbian-config --api help
```

Man page ? If it can be made automatically from the existing.

leggewie, 2025-03-18:
Agreed about help2man.  That's what I was hoping to use.  "armbian-config --help" requires root and IMHO it should not.  Precisely because I want to know what it does before giving it root.

igorpecovnik, 2025-03-18:
> requires root and IMHO it should not.

We are changing system config files, install applications, ... only some parts doesn't need it.

Tearran, 2025-04-02:
@leggewie I agree. 
 During proof of concept, we did some successful man pages. Markdown and Pandoc

This issue is more about organizing relative info 

If you are interested in contributing time to this issue I will contribute what I can to it as well.

Tearran, 2025-04-07:
I  propose closeting this bug report as misleading and misuse of the bug reporting system.  

It's important to note that the absence of a man page is not a bug, but rather a potential enhancement. Using the bug report system for subjective reasons can mislead priorities and misuse the reporting system.

@leggewie I would encourage you to close this bug report and open a feature request with a clear and respectful description of the desired functionality. This approach will ensure it receives the proper attention and consideration.

Your input is valuable, and we appreciate your efforts to improve the project.

Thank you for your understanding.

dimitry-ishenko, 2025-04-14:
I can take a stab at rootless --help switch, if nobody minds.

Another thing I got tripped up on is this:
```
DEBUG=1 armbian-config ...
```
This doesn't work because internally arbian-config will relaunch itself with sudo, but the `DEBUG=1` part (or any other env passed to the command) will get lost.

So, maybe instead of relaunching itself with sudo we should just give an error message saying that armbian-config requires root privileges?

Tearran, 2025-04-14:
>So, maybe instead of relaunching itself with sudo we should just give an error message saying that armbian-config requires root privileges?

This functionality was already in place before, and reverting to it would be taking a step backward. https://github.com/armbian/configng/commit/5ba6109b698c975fae01f83e5e8586dbd541da5c#diff-d9dbde63a26f6a7ff6e27c2efb8c86a666742ab547bf9639d371cdc1789387c4

In my opinion, if a user does not trust the application to run with sudo privileges, they likely should not be using it at all. armbian-config is a system administration tool, and its primary purpose inherently requires administrative access. Investing significant effort to accommodate non-administrators who do not trust the tool to run as root seems counterproductive to its intended use case.

A potential improvement could be relocating the application from bin/ to sbin/, which might further clarify its role as a system administration tool. Instead of treating the DOCUMENTATION.md as a standalone document, it could be used as a structured outline for creating a dedicated man page. This would ensure everything essential—like usage, options, examples, and additional context—is included in a format users are familiar with.

dimitry-ishenko, 2025-04-14:
> In my opinion, if a user does not trust the application to run with sudo privileges, they likely should not be using it at all. armbian-config is a system administration tool, and its primary purpose inherently requires administrative access.

@Tearran sorry, I wasn't trying to imply "trust" or "no trust" issue. The thing I was trying to say is that I got used to using it without sudo, and when I tried to pass an environment variable to it, it took me a some time to figure out what was going on since it wasn't getting through.

Tearran, 2025-04-14:
@dimitry-ishenko My apologies. no intent on implying that. That was inferred here >`Agreed about help2man. That's what I was hoping to use. "armbian-config --help" requires root and IMHO it should not. Precisely because I want to know what it does before giving it root.`

Tearran, 2025-04-14:
@dimitry-ishenko I am Sorry I got distracted. This is not a bug. offered a option and time to a feature request. This is stale and should be closed.

leggewie, 2025-04-20:
Wow, just wow!

I solved the problem for me with "sudo apt purge armbian-config".  All the best!

closed 2025-04-20 by leggewie

Tearran, 2025-04-24:
@leggewie  Truly sad to see the welcomed and constructive discussion go, but it seems this issue was never really about trust or transparency — it was about expecting others to do the work to meet your expectations.

And it’s certainly not about clarity — you asked for more verbosity, yet offered little, instead leaning on vague, unfounded accusations.

This isn’t a critique — it’s entitled hand-waving dressed up as feedback.

leggewie, 2025-04-24:
@Tearran You don't know what you are talking about or what you are doing and that is precisely why giving you root on any of my computers isn't going to happen.  Your lack of knowledge, skill and proper attitude for quality is well documented.

Now, I wish you a good day, Sir.

igorpecovnik, 2025-04-24:
> Your lack of knowledge, skill and proper attitude for quality is well documented.

What we often find ourselves short of is time, patience, understanding, forgiveness, compassion, and love. When these elements align, we're in a much better position to do things well. For now, we're doing the best we can under the circumstances.

Resources are limited—both personally and across the project—and it's completely natural that users have high expectations, especially given the complexity we're dealing with. Managing a fully-fledged operating system, complete with its configuration tools, build framework, documentation, and everything else, is no small task. It’s easy to underestimate the scale of what we’re holding together with so few hands.

When the project eventually has the resources to match the level of expected quality—whether that’s production-grade or beyond—we’ll rise to meet that challenge. I've worked with military-grade software standards before, and while those can be impressive, they’re also often impractical for a community-driven project. Striving for better? Absolutely. But chasing perfection without the means is not only unrealistic—it can be destructive.

We already know where many of the issues lie; there’s no need to point fingers or assign blame. The list of tasks is already longer than we can ever handle, and that’s a reality we've had to come to terms with.

At this stage, compromise is often the only way forward. The alternative is stagnation or burnout. We must also respect that contributors are volunteering their personal time. Our role is to offer guidance and direction—not obligations. Freedom, not pressure, is what keeps this project alive.