armbian/configng #484: [Bug]: removing DE removes network-manager

leggewie, 2025-03-19:
### What happened?

Had a pretty catastrophic experience with armbian-config today.  Background: I installed the Ubuntu Noble XFCE variety because that is the only Ubuntu-Root for Armbian available for the Banana Pi M2+.  I run the board headless and don't need X.  While looking around armbian-config options, I spotted the option to remove XFCE.  This removes network-manager and thus shut me off from the machine.  It also left the system in a bad mess with a lot of broken dependencies in apt.

### How to reproduce?

1. flash XFCE flavor
2. use armbian-config to drop XFCE
3. watch the network go down

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Tearran, 2025-04-02:
Ooh a tuff one. 

Disclaimer: This is a community support board, meaning help comes from volunteers, not guaranteed assistance. So technically…

If this is a one-off project or a documented experiment, I’m happy to help where I can. I have both the M2 Zero and P2 Zero, which at first glance seem to have similar hardware. In my opinion, the most direct solution is to build the image you need using the Armbian build system.

Tearran, 2025-04-02:
@leggewie Welcome, and thank you for your contribution.

Fixing the dependency issues would be valuable. However, to reiterate, the solution is to use the IoT image for IoT/embedded projects. For community-supported devices, the recommended approach is to use the Armbian build system.

If you disagree with this solution may I suggest updating the title to better reflect the dependency issue rather than the current wording. This may be present with something supported.

Would this be something worth bringing to the build developers as a feature request?

leggewie, 2025-04-03:
Thank you for your comment.  Here's my take.

- this isn't going to be board-specific
- the feature is offered and horribly broken, the outcome is quite serious for the user
- armbian is a distribution with package management, not akin to classic IoT/embedded
- I don't think armbian-config should even be doing this, the approach it uses is probably unfixable

igorpecovnik, 2025-04-03:
> the feature is offered and horribly broken

Yes, but lets bring this up properly with APA and simply drop broken functionality. It makes no sense fixing this if it will be deprecated. Hopefully soon.

leggewie, 2025-04-03:
> > the feature is offered and horribly broken
> 
> Yes, but lets bring this up properly with APA and simply drop broken functionality. It makes no sense fixing this if it will be deprecated. Hopefully soon.

100% agreed

closed 2025-04-05 by Tearran