armbian/configng #494: [Bug]: Few issues with the checkpoint behavior

dimitry-ishenko, 2025-03-25:
### What happened?

I am observing a few things with regards to the `checkpoint` and `set_checkpoint` behavior, which I don't quite understand. Specifically, if I run this command:
```bash
admin@uefi-x86:~/configng$ sudo ./bin/armbian-config --help
```
I get the following output:
```bash
Initializing script                     0 seconds
Loaded Runtime variables...             0 seconds
The current Armbian 25.2.3 stable (noble) is supported.           0 seconds
Loaded Dialog...                        0 seconds
Loaded Docs...                          0 seconds
Loaded System helpers...                0 seconds
Loaded Network helpers...               0 seconds
Loaded Software helpers...              0 seconds
Loaded Runtime conditions...            1 seconds
Usage:  armbian-config --[option]
  Options:
     --help [category]   Use [category] to filter specific menu options.
     --cmd  [option]     Run a command from the menu (simple)
     --api  [option]     Run a helper command        (advanced)
     --doc               Generate the README.md file

     Examples:
     armbian-config --help [cmd||System||Software||Network||Localisation]
     armbian-config --cmd help
     armbian-config --api help
```
I get various Loaded messages with timing info. So far so good.

### How to reproduce?

Now, if I run this command:
```bash
admin@uefi-x86:~/configng$ sudo DEBUG=1 ./bin/armbian-config --help
```
The Loaded messages disappear and show up at the end instead:
```bash
Usage:  armbian-config --[option]
  Options:
     --help [category]   Use [category] to filter specific menu options.
     --cmd  [option]     Run a command from the menu (simple)
     --api  [option]     Run a helper command        (advanced)
     --doc               Generate the README.md file

     Examples:
     armbian-config --help [cmd||System||Software||Network||Localisation]
     armbian-config --cmd help
     armbian-config --api help

Total elapsed time            : 1 seconds
Initializing script           : 0 seconds
Loaded Runtime variables...   : 1 seconds
The current Armbian 25.2.3 stable (noble) is supported.: 0 seconds
Loaded Dialog...              : 0 seconds
Loaded Docs...                : 0 seconds
Loaded System helpers...      : 0 seconds
Loaded Network helpers...     : 0 seconds
Loaded Software helpers...    : 0 seconds
Loaded Runtime conditions...  : 0 seconds
DEBUG Messages on             : 0 seconds
```

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dimitry-ishenko, 2025-03-25:
Now, if I run this command (with `DEBUG` set to 1):
```bash
admin@uefi-x86:~/configng$ sudo DEBUG=1 ./bin/armbian-config --help
```
The Loaded messages disappear and show up at the end instead:
```bash
Usage:  armbian-config --[option]
  Options:
     --help [category]   Use [category] to filter specific menu options.
     --cmd  [option]     Run a command from the menu (simple)
     --api  [option]     Run a helper command        (advanced)
     --doc               Generate the README.md file

     Examples:
     armbian-config --help [cmd||System||Software||Network||Localisation]
     armbian-config --cmd help
     armbian-config --api help

Total elapsed time            : 1 seconds
Initializing script           : 0 seconds
Loaded Runtime variables...   : 1 seconds
The current Armbian 25.2.3 stable (noble) is supported.: 0 seconds
Loaded Dialog...              : 0 seconds
Loaded Docs...                : 0 seconds
Loaded System helpers...      : 0 seconds
Loaded Network helpers...     : 0 seconds
Loaded Software helpers...    : 0 seconds
Loaded Runtime conditions...  : 0 seconds
DEBUG Messages on             : 0 seconds
```

This is a bit strange, but OK...

dimitry-ishenko, 2025-03-25:
But, if I run this command (with `DEBUG` set to 2):
```
admin@uefi-x86:~/configng$ sudo DEBUG=2 ./bin/armbian-config --help
```
The Loaded messages show up twice:
```
Initializing script                     0 seconds
Loaded Runtime variables...             0 seconds
The current Armbian 25.2.3 stable (noble) is supported.           0 seconds
Loaded Dialog...                        0 seconds
Loaded Docs...                          0 seconds
Loaded System helpers...                0 seconds
Loaded Network helpers...               0 seconds
Loaded Software helpers...              0 seconds
Loaded Runtime conditions...            0 seconds
DEBUG Messages on                       0 seconds
Usage:  armbian-config --[option]
  Options:
     --help [category]   Use [category] to filter specific menu options.
     --cmd  [option]     Run a command from the menu (simple)
     --api  [option]     Run a helper command        (advanced)
     --doc               Generate the README.md file

     Examples:
     armbian-config --help [cmd||System||Software||Network||Localisation]
     armbian-config --cmd help
     armbian-config --api help

Total elapsed time            : 0 seconds
Initializing script           : 0 seconds
Loaded Runtime variables...   : 0 seconds
The current Armbian 25.2.3 stable (noble) is supported.: 0 seconds
Loaded Dialog...              : 0 seconds
Loaded Docs...                : 0 seconds
Loaded System helpers...      : 0 seconds
Loaded Network helpers...     : 0 seconds
Loaded Software helpers...    : 0 seconds
Loaded Runtime conditions...  : 0 seconds
DEBUG Messages on             : 0 seconds
```

@Tearran Is that the intended behavior?

dimitry-ishenko, 2025-03-25:
Also, pre checkpoint merge I was able to do things like this:

```
admin@uefi-x86:~/configng$ sudo ./bin/armbian-config --api module_nfsd status || echo Not installed
```

But, now I can't because armbian-config always returns exit code 0, even if nfsd is not installed.

What happened is we now have checkpoint commands here:

https://github.com/armbian/configng/blob/262ca6043f2ed94fc2f26378375224d4bc741c9f/bin/armbian-config#L100-L101

and here:

https://github.com/armbian/configng/blob/262ca6043f2ed94fc2f26378375224d4bc741c9f/bin/armbian-config#L115-L116

which hide (overwrite) exit code of the previous command in `--cmd` and `--api`. Not only that, but `checkpoint debug "Blah"` will always set exit code to 1 is `DEBUG` is not defined. It happens here:

https://github.com/armbian/configng/blob/262ca6043f2ed94fc2f26378375224d4bc741c9f/tools/modules/initialize/message_checkpoint.sh#L93-L95

If `DEBUG` is not defined the `[[` command will return 1, which will be propagated by the `case` statement and the `checkpoint` function. So, effectively, `armbian-config` would always return exit code 1 (failure).

@igorpecovnik tried to fix it in a57429264dcf32477f62839c8b4b7f9c2fef3aee but that ended up making `armbian-config` always return 0 exit code, which is what I am observing now.

dimitry-ishenko, 2025-03-25:
What I would like to do is following:

1. Make `checkpoint mark "Blah"` print just the message "Blah" in `UXMODE`. And, if `DEBUG` is defined, it will also print the duration.

2. Make `checkpoint debug "Blah"` print the message and the duration if `DEBUG` is defined (regardless of the `UXMODE`).

3. Make the `checkpoint` function save and restore exit code of the previous command, so that they don't interfere with execution of other commands.

@igorpecovnik @Tearran wanted to get your thoughts on my findings and proposed handling.

Tearran, 2025-03-25:
>@Tearran Is that the intended behavior?

in short yes.  

@dimitry-ishenko  Thanks for the example clarified the issue a lot
```
admin@uefi-x86:~/configng$ sudo ./bin/armbian-config --api module_nfsd status || echo Not installed
```
I am unpacking the rest now :)

Tearran, 2025-03-27:
@dimitry-ishenko has this issue satisfied?

closed 2025-03-27 by dimitry-ishenko