armbian/configng #505: [Bug]: No hardware option in System settings

Robolightning, 2025-04-01:
### What happened?

I can't see Hardware settings in armbian-config "System" on OrangePi 5.

[image]

### How to reproduce?

Install Armbian Ubuntu 24.04 Noble minimal with kernel 6.1+ on OrangePi 5

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Tearran, 2025-04-02:
@Robolightning   

Is it possible you are looking for the "device tree overlay".  for i2c, URT, spi, etc?

`Kernel > SY210   -  Manage device tree overlays`  

[image]

Beyond that this would be more likely a "feature request" 

>NOTE: Not a bug

Robolightning, 2025-04-02:
> [@Robolightning](https://github.com/Robolightning)
> 
> Is it possible you are looking for the "device tree overlay". for i2c, URT, spi, etc?
> 
> `Kernel > SY210 - Manage device tree overlays`
> 
> <img alt="Image" width="584" src="https://private-user-images.githubusercontent.com/2831630/429301883-b9d6beb9-ec7c-489e-b2f3-2d6d32a71c98.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDM1OTIxMzQsIm5iZiI6MTc0MzU5MTgzNCwicGF0aCI6Ii8yODMxNjMwLzQyOTMwMTg4My1iOWQ2YmViOS1lYzdjLTQ4OWUtYjJmMy0yZDZkMzJhNzFjOTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDQwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA0MDJUMTEwMzU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTVlN2IwZDg0MzI1MTVhNjhlOGMxY2IxYzczNDU3MzEzN2Y1ZjMzMDZhOGRkZjgyYTZmMzQyY2RjYzE4YjNmMiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ijTD8swHkvuHAMwlIK3DPOKVzVgsw291lurV3r5rHjo">
> Beyond that this would be more likely a "feature request"
> 
> > NOTE: Not a bug

Ok, thanks, but there is another problem. Only 2 settings available:

[image]

Tearran, 2025-04-02:
I am sorry, but I still fail to see what exactly the issue with armbian-config is. It can only manage what is available. If the hardware feature you need is missing, armbian-config cannot enable or disable it.

I understand your frustration—it can be difficult when something you need isn’t available. However, calling a missing feature a bug is a misunderstanding of how this works. This project relies heavily on volunteers, and as someone who has personally donated hundreds of hours, I find it frustrating when feature requests are mislabeled as bugs.

Here are a few real options to move forward:

Support Armbian by purchasing a platinum-supported board—this directly funds development and helps ensure the features you may need are maintained and improved.

Consider what community support really means—this is a volunteer-driven project, so expectations should align with what the community can reasonably contribute.

Research and share your findings—investigating missing features and sharing insights with the community can be just as valuable as coding, helping guide development efforts.

Without proper manufacturer support, missing features won’t magically appear.

This is my personal perspective as someone who has volunteered significant time to the project and does not necessarily reflect the official views of Armbian. However, Armbian platinum support exists to provide sustainable development, and supporting it benefits the entire ecosystem."

Robolightning, 2025-04-10:
> I am sorry, but I still fail to see what exactly the issue with armbian-config is. It can only manage what is available. If the hardware feature you need is missing, armbian-config cannot enable or [...]

I have investigated the problem. It is not caused by my settings or packages. It happens even with a clean installation of the system (Ubuntu 24 minimal on OrangepPi 5)

closed 2025-04-10 by Robolightning