armbian/configng #520: [Feature Request]: be more verbose (if asked) on what armbian-config is doing

leggewie, 2025-04-05:
### Which feature would you like to have?

armbian-config works as a black box.  It is meant as a tool for the user that can be told to "do XYZ for me" and it will happily do its magic in the back.  It being a black box makes me uneasy (I understand it's open source, that's not the issue).  It would be nice if there was a help knob or some such similar to what you have in kernel makefiles that explains a little bit more what each task does.

### Funding

- [ ] [I am willing to cover expenses](https://www.armbian.com/contact/)

leggewie, 2025-04-05:
https://github.com/armbian/configng/issues/288#issuecomment-2509753624 is an example.  The knob could say what command is being called for the user to inspect, understand and be at ease.

Tearran, 2025-04-05:
@leggewie The magician analogy fits — and I'd take it a step further: you can either trust your magician or start learning the tricks yourself. Wanting both the front-row seat and the backstage pass without getting involved isn’t a reasonable expectation. Transparency isn’t the issue — the code’s open, the documentation exists, and contributors are here to help — but no tool can hand out mastery without effort.

And if you check out `man apt`, you'll see how even well-established tools struggle with this balance. In fact, the maintainers explicitly noted that cramming too much into a single man page led to user confusion, which is why they later split it up. This isn't a trivial problem — it's a usability challenge that requires thoughtful design and active community input.

Expecting both the simplicity of a "do-it-for-me" black box and full internal clarity without investing time or participation isn't realistic. The code is open, the docs are available, and collaboration is welcome — but like any real magic, understanding the trick takes some effort.

(Just sharing my perspective as a volunteer contributor — not on behalf of the Armbian team.)

https://github.com/armbian/configng/edit/main/DOCUMENTATION.md#cli-options would be a good start to make some sugestions

leggewie, 2025-04-20:
OK, after observing the magician for a little bit longer, I have decided I'd rather not trust him.  I was already burned once, you know.  "sudo apt purge armbian-config" it is, then.

closed 2025-04-20 by leggewie