armbian/configng #526: [Bug]: Only two overlays on armbian-config for OrangePi 5 with clean installation

Robolightning, 2025-04-10:
### What happened?

In armbian-config overlays only two positions are displayed:
[ ] opp-oc-24ghz
[ ] panthor-gpu

The rest of the overlays work if they are added via /boot/armbianEnv.txt directly, in the /boot/ folder all the necessary overlays are there. The problem is in armbian-config.
I have already tried several different Armbian Ubuntu 24 images and on the Linux kernel version 6.

It is not caused by my settings or packages. It happens even with a clean installation of the system ([Armbian Ubuntu 24 minimal on OrangepPi 5](https://dl.armbian.com/orangepi5/Noble_vendor_minimal))

[image]

### How to reproduce?

Clean install system ([Armbian Ubuntu 24 minimal](https://dl.armbian.com/orangepi5/Noble_vendor_minimal)) on OrangepPi 5 and go to the armbian-config > System > Kernel > Manage overlays...

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-04-10:
prefixes are wrong. configng is looking for _rockchip-rk3588_ and only finds those two since they have this prefix in file name. All generic overlays start with _rk3588_ only and therefore not visible. Either prefixes should be unified in _build_ or configng adjusted to show these generics too.

Robolightning, 2025-04-10:
> prefixes are wrong. configng is looking for _rockchip-rk3588_ and only finds those two since they have this prefix in file name. All generic overlays start with _rk3588_ only and therefore not visible. Either prefixes should be unified in _build_ or configng adjusted to show these generics too.

Is this a problem on Armbian side, Armbian config or OrangePi/Rockchip developers side?

For my local solution, should I rename the prefixes on the overlays or change the prefix in armbianEnv.txt? Or edit the armbian-config code to add support for both types of prefixes for rk-3588?

Robolightning, 2025-04-10:
> Might be an issue of both Armbian and armbian-config. Former because inconsistent overlay naming and latter for not detecting such 😁 . Jokes aside, not my decision to determine this.
> 
> Anyway, if you don't need armbian-config, simply edit /boot/armbianEnv.txt manually to enable/disable overlays by adjusting `overlays=` to your needs. If you need an example, simply enable like the panthor overlay and check how /boot/armbianEnv.txt looks afterwards. overlay_prefix may need adjustment, depending on used overlays. As for myself I do the not recommended way and rename the dtb files so they match up 🙈

Are you sure that Armbian itself generates overlays during assembly, and does not take them automatically from a third party repository (OrangePi/Rockchip), taking the names "as is"?

EvilOlaf, 2025-04-10:
Depends. In this particular case the _vendor_ kernel source is maintained by Armbian, therefore it is an Armbian issue.
https://github.com/armbian/linux-rockchip

Robolightning, 2025-04-10:
> Depends. In this particular case the _vendor_ kernel source is maintained by Armbian, therefore it is an Armbian issue. https://github.com/armbian/linux-rockchip

Ok, then it is reasonable to create an issue there to rename overlays and overlay_prefix and consider that there is no problem on the armbian-config side?

EvilOlaf, 2025-04-11:
Again, depends. If the consest is to prefix all overlays with _rockchip-rk3588_ then there is no issue in configng. The other way around (prefix all with _rk3588_ only) however there is since those are not detected and that is actually the majority atm.

@amazingfate tl;dr: What's your opinion? Should rk3588 overlays file names start with the prefix _rockchip-rk3588_ or _rk3588_ only? At the moment we have both causing issues.

Tearran, 2025-04-11:
Both topics are duplicates: [#505](https://github.com/armbian/configng/issues/505) and [#523](https://github.com/armbian/configng/issues/523). 

@Robolightning As a reminder from your last issue regarding overlays:

> It possible you are looking for the "device tree overlay". for i2c, URT, spi, etc?  
> Kernel > SY210 - Manage device tree overlays  
> Image Beyond that this would be more likely a "feature request"  
> NOTE: Not a bug  

What level of responsibility are you willing to assume as an image-space user? Are you prepared to make a proper feature request, considering the favor you are asking volunteers to address, or will you continue searching for someone else to take responsibility?

Tearran, 2025-04-11:
>It just seems to me that the problem should also be solved on the maintainers' side

@Robolightning First Thank you for your time and perspective. That toppic is https://github.com/armbian/configng/issues/523.

As i see it the build developers have more pressing matters then to make sure a image-space convenience is made. 

>I solved the problem of the current issue locally, as discussed above, by renaming the overlays and the prefix for the overlays
Again Thank you for your time and experience
.
out of respect for the time you have given 

here is the line you have issue with. 
`if [[ $BOARDFAMILY == "rockchip-rk3588" ]] && [[ $BRANCH == "vendor" ]]; then`
my quick guess at start would be
`if [[ $BOARDFAMILY == "rockchip-rk3588" || $BOARDFAMILY == "rk3588" ]] && [[ $BRANCH == "vendor" ]]; then`

AI Suggested 
```bash
if [[ "$BOARDFAMILY" =~ ^(rockchip-rk3588|rk3588)$ && "$BRANCH" == "vendor" ]]; then
    # Add support for rk3588 vendor kernel overlays
    # Place your logic here
fi
```

How you want to prosed is up to you! Do you want to continue to burden the developers with things tinkers, hobbyist. and AI can do or help with the  prefix issue?

Tearran, 2025-04-11:
@Robolightning Sorry timing was off there https://github.com/armbian/configng/pull/528