armbian/configng #541: [Bug]: Invalid overlay_prefix

sp2ong, 2025-04-21:
### What happened?

BOARD=orangepizero3
BOARD_NAME="Orange Pi Zero3"
BOARDFAMILY=sun50iw9
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=76229701c
LINUXFAMILY=sunxi64
ARCH=arm64
BOOT_SOC=
IMAGE_TYPE=nightly
BOARD_TYPE=csc
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
KERNEL_TARGET=current,edge
KERNEL_TEST_TARGET=current
FORCE_BOOTSCRIPT_UPDATE=yes
FORCE_UBOOT_UPDATE=
OVERLAY_DIR="/boot/dtb/allwinner/overlay"
VENDOR="Armbian"
VENDORCOLOR="247;16;0"
VENDORDOCS="https://docs.armbian.com"
VENDORURL="https://www.armbian.com/"
VENDORSUPPORT="https://forum.armbian.com"
VENDORBUGS="https://www.armbian.com/bugs"
BOOTSCRIPT_FORCE_UPDATE="yes"
BOOTSCRIPT_DST="boot.cmd"
VERSION=25.5.0-trunk.426
REVISION=25.5.0-trunk.426
BRANCH=current

uname -a
Linux armbian 6.12.23-current-sunxi64 #1 SMP Thu Apr 10 12:39:41 UTC 2025 aarch64 GNU/Linux

 I noticed that after updating the kernel to version 6.12.23 there is a problem with armbian-config 

I can't set overlay tree devices choosing from 

**armbian-config --cmd SY210** 

The current Armbian 25.5.0-trunk.426 nightly (bookworm) is supported. 
Check distro 
Loaded Dialog... 
Loaded Docs... 
Loaded System helpers... 
Loaded Network helpers... 
Loaded Software helpers... 
Loaded Runtime conditions... 
Invalid overlay_prefix sun50i-h616  

### How to reproduce?

run command on Orange Pi Zero 3: 

armbian-config --cmd SY210

https://paste.armbian.com/ipahimocir

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ZjemCiKolege, 2025-04-21:
+ in uboot

https://i.imgur.com/og3jGOj.png

sp2ong, 2025-04-21:
OK I have updated armbian-config to latest version and look now show devicetree

[image]

closed 2025-04-21 by sp2ong

sp2ong, 2025-04-22:
> * uboot log after update to 426
> 
> 
> https://i.imgur.com/og3jGOj.png [5bf53cf#commitcomment-155652292](https://github.com/armbian/configng/commit/5bf53cf1fc01290c1ca5c4d779e08aba42e79bfc#commitcomment-155652292)
> 
> looks like armbian-config adds sun50i-h616 to gpu/uart5 etc

Yes, I confirm when I remove in amrbianEnv.txt in overlays=  prefix sun50i-h616 from name sun50i-h616-uart5 and now
overlays=uart5 
I have /dev/ttyS1 in my system. 
But if anybody make upgrade system from 6.6.75 kernel to 6.12.23 the old name UART5 was uart5-ph it will remove access in system to /dev/ttyS1.  I don't know what mean -ph" in name but I see still in device tree exist -ph -pi -pg