armbian/configng #543: [Bug]: Armbian-config Net Problem

danielmendoncabr, 2025-04-26:
### What happened?

i've compiled my own .img with Armbian Build (i think official img there is no problem)
 v25.05 rolling for Orange Pi PC + running Armbian Linux 6.12.23-current-sunxi
Debian rolling (bullseye)
and when trying to edit Network with Armbian-config the follow error occurs

`Traceback (most recent call last):
  File "/usr/sbin/netplan", line 23, in <module>
    netplan.main()
  File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 257, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 48, in run
    self.run_command()
  File "/usr/share/netplan/netplan/cli/utils.py", line 257, in run_command
    self.func()
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 58, in command_set
    self.write_file(set_tree, self.origin_hint + '.yaml', self.root_dir)
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 108, in write_file
    new_tree = self.merge(config, set_tree)
  File "/usr/share/netplan/netplan/cli/commands/set.py", line 85, in merge
    if key in a:
TypeError: argument of type 'NoneType' is not iterable`

How to fix:
`sudo nano /usr/share/netplan/netplan/cli/commands/set.py`

add
`import shutil`

and search for 
`os.replace(tmpp, absp)`

and change for
`shutil.move(tmpp, absp)`

if still doesn't work edit 
`sudo nano /etc/netplan/armbian.yaml`

erase everything and paste
because i cannot format here every # means 1 space click, ok? '-'

network:
##version: 2
##renderer: networkd
##ethernets:
####eth0:
######dhcp4: true

(if still doesn't work, erase everything in /etc/netplan/ [must be two *.yaml files] and create a new armbian.yaml with the code above)

### How to reproduce?

open armbian-config > Network > Basic Network Setup > Anything from here

### Branch

other (only relevant in corner cases)

### On which host OS are you running the build script and observing this problem?

Other

### Are you building on Windows WSL2?

- [ ] Yes, my Ubuntu/Debian/OtherOS is running on WSL2

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

EvilOlaf, 2025-04-26:
moved to configng since not a framework issue.

igorpecovnik, 2025-04-26:
> Debian rolling (bullseye)

I think this also not an armbian-config issues but bug / too old version of NetPlan that comes with bullseye. We just declared end of support for those https://github.com/armbian/build/commit/3af435415197d63fce72a0d89c9ba0ea07043fdc

There are too much of problems with current stable OS versions (Bookworm / Noble). First install that, try again and report if that solves the issue.

Tearran, 2025-04-26:
I previously added a JSON entry where nmtui could be integrated with configng in the same way a module would be. Would it be worth adding this to the menu for basic use?

Also some reusable ui module may be useful for Netplan cli wrappers #

igorpecovnik, 2025-04-26:
> Would it be worth adding this to the menu for basic use?

It won't fix this problem here. Older versions of Debian has too old version of NetPlan, which is likely causing those problems and here we can't do much as recommending upgrade to / using current stable OS. Supporting current user-space is already on the edge of possible, while older variants is simply out of sane range.

igorpecovnik, 2025-05-05:
I tried to re-create this on Bananapi PRO, A20, which is using same kernel. Logs: https://paste.armbian.com/iketexubuk

- setting fixed IP on LAN
- setting DHCP on wireless
- removing fall back configuration

Armbian bookworm, build from sources. 

No issues found, closing.

closed 2025-05-05 by igorpecovnik