armbian/configng #592: [Bug]: armbian-config does not takes care about

Race666, 2025-05-04:
### What happened?

I installed armbian on a Orange Pi Zero and have enabled the uart 1 and 2 overlay by armbian.

```
 [*] sun8i-h3-uart1
 [*] sun8i-h3-uart2             
```               

This does not work because in /boot/armbianEnv.txt a preconfigured prefix is set

`overlay_prefix=sun8i-h3`

armbian-config sets 

`overlays=sun8i-h3-uart1 sun8i-h3-uart2 `

and this causes that the load of the overlays fails because the loader searches for 

```
/boot/dtb/overlay/sun8i-h3-sun8i-h3-uart1.dtbo
/boot/dtb/overlay/sun8i-h3-sun8i-h3-uart2.dtbo

```
instead of 
```

/boot/dtb/overlay/sun8i-h3-uart1.dtbo
/boot/dtb/overlay/sun8i-h3-uart1.dtbo
```

### How to reproduce?

- Install https://www.armbian.com/orange-pi-zero/
- start armbian-config
- enable overlays sun8i-h3-uart1 and sun8i-h3-uart2

this results in a /boot/armbianEnv.txt

```
verbosity=1
bootlogo=false
console=both
disp_mode=1920x1080p60
overlay_prefix=sun8i-h3
overlays=sun8i-h3-uart1 sun8i-h3-uart2
rootdev=UUID=aec93845-9ef3-416e-ba6a-9f6e099bce15
rootfstype=ext4
...
```

and the path for loading the overlays is like this:

/boot/dtb/overlay/$OVERLAYPREFIX/$OVERLAY

if I set
```
verbosity=1
bootlogo=false
console=both
disp_mode=1920x1080p60
overlay_prefix=sun8i-h3
overlays=uart1 uart2
rootdev=UUID=aec93845-9ef3-416e-ba6a-9f6e099bce15
rootfstype=ext4
...
```
it works as expected.

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Debian 12 Bookworm

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

sp2ong, 2025-05-10:
It was solved 

https://github.com/armbian/configng/issues/344

but It looks like bug with overlay_prefix back again

TMSL, 2025-06-09:
I'm running:
```
BOARD=renegade
BOARD_NAME="Renegade"
BOARDFAMILY=rockchip64
BUILD_REPOSITORY_URL=https://github.com/armbian/build
BUILD_REPOSITORY_COMMIT=ffe959bb1
LINUXFAMILY=rockchip64
ARCH=arm64
BOOT_SOC=rk3328
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
KERNEL_TARGET=current,edge
KERNEL_TEST_TARGET=current
FORCE_BOOTSCRIPT_UPDATE=
FORCE_UBOOT_UPDATE=
OVERLAY_DIR="/boot/dtb/rockchip/overlay"
VENDOR="Armbian"
VENDORDOCS="https://docs.armbian.com"
VENDORURL="https://www.armbian.com"
VENDORSUPPORT="https://forum.armbian.com"
VENDORBUGS="https://www.armbian.com/bugs"
BOOTSCRIPT_FORCE_UPDATE="no"
BOOTSCRIPT_DST="boot.cmd"
VERSION=25.2.2
REVISION=25.2.2
```

closed 2025-07-12 by igorpecovnik