armbian/configng #626: [Bug]: when i set the wifi ,this error occured.and failed setting the wifi

HeibaiPanda, 2025-07-03:
### What happened?

Traceback (most recent call last):
File "/usr/sbin/netplan", line 23, in
netplan.main()
File "/usr/share/netplan/netplan/cli/core.py", line 50, in main
self.run_command()
File "/usr/share/netplan/netplan/cli/utils.py", line 257, in run_command
self.func()
File "/usr/share/netplan/netplan/cli/commands/set.py", line 48, in run
self.run_command()
File "/usr/share/netplan/netplan/cli/utils.py", line 257, in run_command
self.func()
File "/usr/share/netplan/netplan/cli/commands/set.py", line 58, in command_set
self.write_file(set_tree, self.origin_hint + '.yaml', self.root_dir)
File "/usr/share/netplan/netplan/cli/commands/set.py", line 108, in write_file
new_tree = self.merge(config, set_tree)
File "/usr/share/netplan/netplan/cli/commands/set.py", line 85, in merge
if key in a:
TypeError: argument of type 'NoneType' is not iterable

### How to reproduce?

trying to connect a wifi

### On which OS are you observing this problem? (for easier diagnosis please provide logs with `armbianmonitor -u`)

Armbian Ubuntu 24.04 Noble

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

HeibaiPanda, 2025-07-25:
I used the command armbian-config and failed to connect a wifi. This is how i set the wifi, and what it displayed to me.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

when i quitted the command armbian-config. the error information showed.

[image]

igorpecovnik, 2025-07-26:
Ah, you are not running Armbian, but a fork, fake Armbian with changes that I am unaware of ... and bullseye variant, which doesn't have good support for Netplan.

1. Any (official) supported Armbian
2. Debian Bookworm or more recent
3. Ubuntu Jammy or more recent

This could be related to Netplan package that comes with Bullseye ... blind guess.